adafruit/RTClib #37: Line 59

Apurvap24, 2015-03-04:
When I compile it says:

In file included from Home_security.ino:9:
C:\Users\Apurva\Documents\Arduino 1.06\arduino-1.0.6\libraries\RTClib/RTClib.h:59: error: expected identifier before '{' token
C:\Users\Apurva\Documents\Arduino 1.06\arduino-1.0.6\libraries\RTClib/RTClib.h:59: error: expected `}' before '{' token
C:\Users\Apurva\Documents\Arduino 1.06\arduino-1.0.6\libraries\RTClib/RTClib.h:59: error: expected unqualified-id before '{' token
C:\Users\Apurva\Documents\Arduino 1.06\arduino-1.0.6\libraries\RTClib/RTClib.h:60: error: expected declaration before '}' token

ladyada, 2015-10-06:
need more info, what board settings, did the built in examples work, etc? seems to be fine for me. plz reopen if you have more details. thx!

closed 2015-10-06 by ladyada

Apurvap24, 2015-10-06:
Sorry I was think something else than, but than it worked perfectfly for me.

tnsolley, 2019-02-13:
I've actually got a similar error; when compiling using the 4Duino and 4D Systems software, I get;
```
C:\Program Files (x86)\Arduino\libraries\RTClib/RTClib.h:79:25: note: in expansion of macro 'OFF'

 enum Ds1307SqwPinMode { OFF = 0x00, ON = 0x80, SquareWave1HZ = 0x10, SquareWave4kHz = 0x11, SquareWave8kHz = 0x12, SquareWave32kHz = 0x13 };

                         ^

In file included from H:\Dev\Cosplay_pictures\Legacy_Mando_SWEU\Gauntlet_Display\Gauntlet_Display\GD_ver3\GD_ver3.ino:26:0:

C:\Program Files (x86)\Arduino\libraries\RTClib/RTClib.h:79:139: error: expected declaration before '}' token

 enum Ds1307SqwPinMode { OFF = 0x00, ON = 0x80, SquareWave1HZ = 0x10, SquareWave4kHz = 0x11, SquareWave8kHz = 0x12, SquareWave32kHz = 0x13 };

                                                                                                                                           ^
```