adafruit/RTClib #43: How do you set date and time?

liudr, 2015-07-18:
With the old code that I've got before you guys started going with github, DateTime has simple inline functions to set up date and time:

    void set_year(int y) { yOff=y-2000;}
    void set_month(int y) { m=y;}
    void set_day(int y) { d=y;}
    void set_hour(int y) { hh=y;}
    void set_minute(int y) { mm=y;}
    void set_second(int y) { ss=y;}

Now they are all removed, how am I supposed to set a DateTime object to a particular date and time? The only option I see is to construct one with the constructor, and then there is no way to change it. Am I missing something obvious?

louwin, 2015-07-18:
A newbie here but I think you use the rtc.adjust method

liudr, 2015-07-18:
Yes miss(ter)! But that way you have to create a new DateTime instance every time you want an RTC adjustment or time calculation. What I would like to see is having ONE DateTime instance. Set its date and time to something at some point in program and use rtc.adjust to adjust the RTC. Then later if you need another adjustment or some calculation of future time for comparison, load it with different date time values. This way we have a smaller memory footprint. One instance of DateTime in memory at all time compared with a bunch of integers/bytes for yy,mm,dd,hh,mn,ss and in addition, a DateTime so the numbers can initialize it.

ladyada, 2015-10-06:
yah i dont know where or how that left. if you want to submit a chunk of code for manually setting a DateTime after creation, thats cool by me, ill merge :)

closed 2015-10-06 by ladyada