adafruit/RTClib #51: Merge for DS3234/DS3231 support

awickert, 2015-10-27:
Howdy! I wrote code to give good support for the DS3234 and DS3231 a few years ago, but it's in a file structure that's significantly different than what you have here. I think it would be nice to send all this work back your way to have it in the main branch that most people use; could you provide some advice on how to do this?

DS3234 is a more direct fork of an old version of RTClib:
https://github.com/NorthernWidget/RTClib

DS3231 is more its own beast, but includes the DateTime object, as inspired by RTClib:
https://github.com/NorthernWidget/DS3231

Would love to get all of this in one place -- let me know if you have ideas/guidelines for how you'd like to see RTClib evolve.

MrAlvin, 2016-01-21:
You might find it interesting to look at this https://github.com/MrAlvin/RTClib
I have combined/merged basic DS3231 code into the code structure that the Adafriut library uses, and have successfully tested alarm1 functionality.

awickert, 2016-02-22:
Thanks @MrAlvin -- I'm going to check that out!

drak7, 2019-07-01:
It's been a while but if you'd still like to add DS3234 support feel free to submit a PR!

closed 2019-07-01 by drak7