adafruit/RTClib #64: ::begin() always return true

av1024, 2016-04-07:
There is no way to check if RTC chip is not available because `begin ()` simply returns `true`.

On my sensor board I use workaround for DS3231 but similar code should work on other IC:

```c
bool RTC_DS3231::begin(void) {
  Wire.begin();
  uint8_t x = read_i2c_register(DS3231_ADDRESS, 6); // 0x00..0x99 - Year in BCD
  return x != 0xFF; // got 0xFF if chip not present
}
```

gutierrezps, 2016-04-17:
I agree. I was facing issues with bad wire hookup on my protoboard, and I only knew the RTC wasn't properly connected because rtc.now() returned a weird date, like "2165/165/165".

Other approach would be to check if endTransmission() == 3 (Addr NACK) for a dummy byte, as stated in [Wire reference](https://www.arduino.cc/en/Reference/WireEndTransmission)

Talking about the RTC_XXX::begin() member, I would remove Wire.begin() from it, since I'm using ESP8266, and I could assign other pins than the defaults to I2C.

av1024, 2016-04-18:
@gutierrezps +1!

I also use this lib with ESP8266, but with "standard" I2C pins so calling Wire.begin() does not matter. Also (at least at esp8266 library 2.1.0) You may call `Wire.begin(sda, scl) once in your code and pin assignments will preserved in next `.begin()` calls.

About endTransmission(): I'm not sure about valid result codes in esp8266 implementation (looks like `twi_writeTo` returns 3 as NACK on transmit data and 2 as NACK address). 

PS: Got "Bad gateway" on Wire reference url :(

bhelterline, 2016-08-01:
+1 I modified my version to watch the result of endTransmission() and added a write in begin() so it actually returns false when there is no RTC.

```
boolean RTC_DS1307::begin(void) {
  Wire.begin();
  Wire.beginTransmission(DS1307_ADDRESS);
  Wire._I2C_WRITE((byte)0);
  if ( Wire.endTransmission() == 0 ) {
  return true;  
  }
  return false;
}

```

ortegafernando, 2016-12-11:
Hi, is there any plan to include this in the master brand of the library ???? Thanks

mouridis, 2019-10-23:
I submitted a PR for this issue guys but it only covers DS3231 which I have and I am able to confirm that the fix works without issues under AVR, ESP8266 and ESP32. Hopefully, it will be merged.

PR #166 (2020-05-13): Verify RTC communication in PCF8523 and DS1307 begin()
This change extends the logic already used for verifying comms with the DS3231 (ff898af) to the other two RTCs supported by this library: the PCF8523 and DS1307.

Fixes #64

I tested these controllers:
Feather M4 Express
Feather nRF52840 Sense
Arduino Mega

with these RTCs:
DS1307  DIY breakout board kit modified for pullup to 5v or 3v.
PCF8523 on RTC + SD Adalogger FeatherWing

Test Note: When `rtc.begin()` returns false, most examples print "Couldn't find RTC", and immediately go into an infinite `while(1)` loop. On the nRF52840 Sense, u [... cut]
commit 3f379ab1 (2020-05-13): Verify RTC communication in PCF8523 and DS1307 begin()

This change extends the logic already used for verifying comms with the DS3231 (ff898af) to the other two RTCs supported by this library: the PCF8523 and DS1307.

Fixes #64
closed 2020-05-13 by drak7