adafruit/RTClib #66: Include a license

FuzzyWuzzie, 2016-05-31:
Right now, this repository doesn't seem to have a defined license, which means it is under the full copyright of Adafruit, which means:

> [Adafruit] retains all rights to [the] source code and that nobody else may reproduce, distribute, or create derivative works from [the] work. (http://choosealicense.com/no-license/)

By not allowing the creation of derivative works, the library on its own is not really useful—I would like to use this library in a project, but unfortunately cannot.

For help choosing a license, I would use http://choosealicense.com/. Personally I would use a BSD, MIT, or Apache 2.0 license, but the choice is completely up to Adafruit and I fully support Adafruit choosing whichever license they desire (so long as they explicitly mention it, even if it is a full copyright).

Thanks!

GrandNewbien, 2017-07-13:
I mean, if it's included as a library in other projects, you shouldn't be including this code with it, just point it here.

edgar-bonet, 2018-09-19:
@ladyada: It would be nice if people could legally use the library, or at least be clearly told that they can't.

ladyada, 2018-09-19:
https://github.com/adafruit/RTClib/blob/master/RTClib.cpp#L2

closed 2018-09-19 by ladyada

massimiliano-cannata, 2018-09-26:
// Released to the public domain! Enjoy! 
It is not really a license !!!! Can you provide a better indication of the type of license? GPLv2?
Other Adafriut lib have license selected in github, just create a new file "license.md" and select your preferred one....