adafruit/RTClib #71: DS3231 not correctly reading date

tchilton, 2016-11-20:
According to the Maxim data sheet, the DS3231 has a century bit in BIT 7 of the 5th register. The current code does not mask this out, which means that the returned date is incorrect (2023 returned for 2016).  See the .c file line 471 which reads 
uint8_t d = bcd2bin(Wire._I2C_READ());

The same masking should be applied as is done in the seconds register
uint8_t d = bcd2bin(Wire._I2C_READ() & 0x7F);

Presumably the century flag should be incorporated somewhere else too ?? I didn't look at this since I dont need it at the moment.

This problem is visible if you hook up an Arduino to a Raspberry Pi on the same I2C bus, then have the RPi obtain time via NTP, then do a "hwclock -w" to write the current time to the RTC module.  Its not clear why the RPi sets the century flag given that we are in 2016, but that's a different issue !!
You can also demonstrate the same point by clearing the month bit with something like  
i2cset -f -y 1 0x68 0x05 0x11
However the RPi will reset the bit to 0x91 when the next "hwclock -w" is performed

TD-er, 2021-07-20:
This issue is still present.
The PR that tried to fix it was closed because some other issues fixed in there were dealt with by another PR.

See datasheet:
[image]

If the module was set on another device, it might have the month set to some value > 12.

Also strictly speaking if the module is set using this library, and read out via other code, it may show the wrong century.

Something similar can happen to the hours as the module can be set to AM/PM mode.

From the datasheet:

> The DS3231 can be run in either 12-hour or 24-hour mode. Bit 6 of the hours register is defined as the 12- or 24-hour mode select bit. When high, the 12-hour mode is selected. In the 12-hour mode, bit 5 is the AM/PM bit with logic-high being PM. In the 24-hour mode, bit 5 is the 20-hour bit (20–23 hours). The century bit (bit 7 of the month register) is toggled when the years register overflows from 99 to 00.

Also not sure what will happen to the DOW if the century is wrong.
I doubt there is a check in the chip to recompute the DOW, but it might be set using another library.

caternuson, 2021-11-02:
Added mask for the century bit in #246 . Hopefully fixes this.

closed 2021-11-02 by caternuson