adafruit/RTClib #73: Code not working with Arduino Due

scumulder, 2016-12-05:
- Arduino board:  Arduino Due

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.6.12

- I'm using an old datalogger shield with my Arduino Due. I am unable to set the current time on the RTC using rtc.adjust(DateTime(F(__DATE__), F(__TIME__)));

My serial monitor keeps showing the following output:

[image]

cwt137, 2017-06-03:
I get the same output when using a PCF8523 and an Arduino UNO R3 compatible device. Am I doing something wrong? I'm new to Arduino programming and building circuits.

melucas, 2018-04-27:
I'm getting the same message "RTC is not running" and I can not set the time on the board.  

I have an Arduino UNO R3 board connected to a ZS-042 clock shield.

Clock SDA connected to Pin 20 on UNO
Clock SCL connected to Pin 21 on UNO

Any suggestions on how to resolve this?

Spomky, 2018-10-26:
Hi @scumulder,

I have the same issue with my Arduino Due board.
Did you fixed it? If yes how did you managed that?

Regards

melucas, 2018-10-29:
Hi @Spomky,

I got it working using the SDA1 and SCL1 pins on my  DUE board.

messmerd, 2019-05-06:
I received the same error with the Due, so I commented out the line: 
`#define Wire Wire1`
in the library's .cpp file, and wired it up to use SDA and SCL, and it worked fine. 
I think I also had to remove the button cell battery of my RTC to make it lose power, then put it back in.

drak7, 2019-10-09:
Sounds like a wiring issue, the library uses Wire1 (SDA1/SCL1) on Due by default.

closed 2019-10-09 by drak7