adafruit/RTClib #97: [SOLUTION] Arduino MKR GSM 1400 incompatability

jgartner1, 2018-08-01:
When using MKRGSM.h in combination with RTClib.h the compiler throws an error because of a redeclaration of the constant 'OFF'. 

The original code in RTClib.h is: 
`enum Ds1307SqwPinMode { OFF = 0x00, ON = 0x80, SquareWave1HZ = 0x10, SquareWave4kHz = 0x11, SquareWave8kHz = 0x12, SquareWave32kHz = 0x13 };`

And I changed it up to: 
`enum Ds1307SqwPinMode { RTC_OFF = 0x00, ON = 0x80, SquareWave1HZ = 0x10, SquareWave4kHz = 0x11, SquareWave8kHz = 0x12, SquareWave32kHz = 0x13 };`

Please consider implementing this solution into your github libary to ensure compatability with the MKR GSM 1400.

jgartner1, 2018-08-01:
Yeah you cloud also do that, but exesting code would have to be changed in your soultion too. But let's face it: Who updates a libary with such predefined tasks? It can do everything it should and security patches aren't necessary. So I think you can change it without destroying a lot of older code, but you make it easier for a lot of new people who download the libary

drak7, 2019-07-16:
Fixed in #116.

closed 2019-07-16 by drak7