adafruit/RTClib #98: ESP32

rahmaevao, 2018-08-17:
For remaping pins SDA and SCL correct file RTClib.cpp:

https://github.com/adafruit/RTClib/blob/82d00a1a15bcf738b001e42bf26a3d1f1a15be32/RTClib.cpp#L240

Use:
Wire.begin(RTC_SDA, RTC_SCL);

lemonbuzz, 2018-11-12:
I have an ESP32 project and I cant seem to get my DS3231 to work....I've remapped the SDA and SCL pins as you suggest here but to no avail

rahmaevao, 2018-11-13:
Remaping can only help configuration I2C in ESP32.
Problem can be in DC3231, in wire or another.
Can you view the SDA and SLC line with an oscilloscope or logic analyzer?

cbpdk, 2019-03-15:
I have just tested a RTClib, DC3231, with a LoadLin32 Lite clone which does not have pin 21.
To make it work for DS3231, I made the following changes in the RTClib.cpp:

////////////////////////////////////////////////////////////////////////////////
// RTC_DS3231 implementation

#if defined( ESP32 ) || defined(ARDUINO_ARCH_ESP32) 
boolean RTC_DS3231::begin(int sda, int scl) {
  Wire.begin(sda,scl);
  return true;
}
#else
boolean RTC_DS3231::begin(void) {
  Wire.begin();
  return true;
}
#endif

And in the RTClib.h:

class RTC_DS3231 {
public:
#if defined( ESP32 ) || defined(ARDUINO_ARCH_ESP32) 
    boolean begin(int sda = 21, int scl = 22);
#else
    boolean begin(void);
#endif
    static void adjust(const DateTime& dt);:

And the DS3231 works with RTClib.

closed 2019-03-21 by rahmaevao