adafruit/RTClib #99: Warning when compiling

MarioJose, 2018-08-28:
Code working, but get warning when compiling

/home/user/.platformio/lib/RTClib_ID83/RTClib.cpp: In constructor 'DateTime::DateTime(uint32_t)':
/home/user/.platformio/lib/RTClib_ID83/RTClib.cpp:86:18: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
if (days < 365 + leap)
^

Phoenix1747, 2018-08-28:
Yeah that's just because you compare the ``uint16_t days` to a normal signed integer `365`. This shouldn't be a real problem at all. By the way, I don't get this warning, how do you use this function?

MarioJose, 2018-09-01:
My code:

```{c}
#include <Arduino.h>

#include <string.h>
#include <inttypes.h>

#include <SD.h>
#include <RTClib.h>

RTC_DS1307 rtc;
DateTime calendar;

void setup()
{
  Serial.begin(9600);

  calendar = rtc.now();

  Serial.println(calendar.minute());

}

void loop()
{

}

```

And Atom+PlatformIO build output:

```
Processing uno (platform: atmelavr; board: uno; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Atmel AVR > Arduino Uno
SYSTEM: ATMEGA328P 16MHz 2KB RAM (31.50KB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf

LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <RTClib> 1.2.1
|   |-- <Wire> 1.0
|-- <SD> 1.2.2
|   |-- <SPI> 1.0
Compiling .pioenvs/uno/src/main.cpp.o
Compiling .pioenvs/uno/lib741/Wire/Wire.cpp.o
Compiling .pioenvs/uno/lib741/Wire/utility/twi.c.o
Compiling .pioenvs/uno/lib80d/RTClib_ID83/RTClib.cpp.o
Compiling .pioenvs/uno/lib234/SPI/SPI.cpp.o
/home/user/.platformio/lib/RTClib_ID83/RTClib.cpp: In constructor 'DateTime::DateTime(uint32_t)':
/home/user/.platformio/lib/RTClib_ID83/RTClib.cpp:86:18: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
if (days < 365 + leap)

^
Compiling .pioenvs/uno/libdc1/SD_ID868/File.cpp.o
Archiving .pioenvs/uno/lib741/libWire.a
Archiving .pioenvs/uno/lib80d/libRTClib_ID83.a
Indexing .pioenvs/uno/lib741/libWire.a
Indexing .pioenvs/uno/lib80d/libRTClib_ID83.a
Compiling .pioenvs/uno/libdc1/SD_ID868/SD.cpp.o
Archiving .pioenvs/uno/lib234/libSPI.a
Compiling .pioenvs/uno/libdc1/SD_ID868/utility/Sd2Card.cpp.o
Compiling .pioenvs/uno/libdc1/SD_ID868/utility/SdFile.cpp.o
Indexing .pioenvs/uno/lib234/libSPI.a
Compiling .pioenvs/uno/libdc1/SD_ID868/utility/SdVolume.cpp.o
Archiving .pioenvs/uno/libFrameworkArduinoVariant.a
Indexing .pioenvs/uno/libFrameworkArduinoVariant.a
Compiling .pioenvs/uno/FrameworkArduino/CDC.cpp.o
Compiling .pioenvs/uno/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pioenvs/uno/FrameworkArduino/HardwareSerial0.cpp.o
Archiving .pioenvs/uno/libdc1/libSD_ID868.a
Indexing .pioenvs/uno/libdc1/libSD_ID868.a

Compiling .pioenvs/uno/FrameworkArduino/HardwareSerial1.cpp.o
Compiling .pioenvs/uno/FrameworkArduino/HardwareSerial2.cpp.o
Compiling .pioenvs/uno/FrameworkArduino/HardwareSerial3.cpp.o
Compiling .pioenvs/uno/FrameworkArduino/IPAddress.cpp.o
Compiling .pioenvs/uno/FrameworkArduino/PluggableUSB.cpp.o
Compiling .pioenvs/uno/FrameworkArduino/Print.cpp.o
Compiling .pioenvs/uno/FrameworkArduino/Stream.cpp.o
Compiling .pioenvs/uno/FrameworkArduino/Tone.cpp.o
Compiling .pioenvs/uno/FrameworkArduino/USBCore.cpp.o
Compiling .pioenvs/uno/FrameworkArduino/WInterrupts.c.o
Compiling .pioenvs/uno/FrameworkArduino/WMath.cpp.o
Compiling .pioenvs/uno/FrameworkArduino/WString.cpp.o
Compiling .pioenvs/uno/FrameworkArduino/abi.cpp.o
Compiling .pioenvs/uno/FrameworkArduino/hooks.c.o
Compiling .pioenvs/uno/FrameworkArduino/main.cpp.o
Compiling .pioenvs/uno/FrameworkArduino/new.cpp.o
Compiling .pioenvs/uno/FrameworkArduino/wiring.c.o
Compiling .pioenvs/uno/FrameworkArduino/wiring_analog.c.o

Compiling .pioenvs/uno/FrameworkArduino/wiring_digital.c.o
Compiling .pioenvs/uno/FrameworkArduino/wiring_pulse.S.o
Compiling .pioenvs/uno/FrameworkArduino/wiring_pulse.c.o
Compiling .pioenvs/uno/FrameworkArduino/wiring_shift.c.o
Archiving .pioenvs/uno/libFrameworkArduino.a
Indexing .pioenvs/uno/libFrameworkArduino.a
Linking .pioenvs/uno/firmware.elf
Checking size .pioenvs/uno/firmware.elf
Building .pioenvs/uno/firmware.hex
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [==        ]  19.7% (used 404 bytes from 2048 bytes)
PROGRAM: [=         ]  10.5% (used 3380 bytes from 32256 bytes)
========================= [SUCCESS] Took 2.56 seconds =========================
```

edgar-bonet, 2018-09-01:
The fix is pretty simple: make `365` unsigned.

```diff
diff --git a/RTClib.cpp b/RTClib.cpp
index a08dfe9..70c5481 100644
--- a/RTClib.cpp
+++ b/RTClib.cpp
@@ -83,9 +83,9 @@ DateTime::DateTime (uint32_t t) {
     uint8_t leap;
     for (yOff = 0; ; ++yOff) {
         leap = yOff % 4 == 0;
-        if (days < 365 + leap)
+        if (days < 365U + leap)
             break;
-        days -= 365 + leap;
+        days -= 365U + leap;
     }
     for (m = 1; ; ++m) {
         uint8_t daysPerMonth = pgm_read_byte(daysInMonth + m - 1);
```

I would happily submit this as a pull request, but it seems the maintainers of this repo have not responded to pull requests since late 2015...

ondras12345, 2020-01-13:
> By the way, I don't get this warning, how do you use this function?

Arduino IDE doesn't show compiler warnings by default...

Phoenix1747, 2020-01-13:
True, I didn't know that back then. Only had verbose compilation output on not the warnings.

commit 13ac6145 (2020-04-21): Fix signed/unsigned compare warnings

This fixes two gcc warnings about "comparison between signed and
unsigned integer expressions".

Fixes #99.
closed 2020-04-21 by drak7