adafruit/RTClib #114: DateTime corruption?

lovethat777, 2019-06-18:
DateTime causes problems.
The following code is self-explanatory: when _ct_ is declared, the _day_ and _month_ return invalid values:

```#include "RTClib.h"      // https://github.com/adafruit/RTClib
#include <Time.h>        // https://github.com/PaulStoffregen/Time
#include <Timezone.h>    // https://github.com/JChristensen/Timezone

// If you declare "DateTime ct;", both month (13) and day (140) are invalid:
// 1560889639 2019-13-140 20:27:19

// If you UNCOMMENT "DateTime ct;" you get the following, which is correct:
// 1560889639 2019-6-18 20:27:19
//DateTime ct;

void setup () {
  Serial.begin(115200);

  time_t nowTime = 1560889639;

  Serial.print(nowTime);
  Serial.print(" ");
  Serial.print(year(nowTime));
  Serial.print("-");
  Serial.print(month(nowTime));
  Serial.print("-");
  Serial.print(day(nowTime));
  Serial.print(" ");
  Serial.print(hour(nowTime));
  Serial.print(":");
  Serial.print(minute(nowTime));
  Serial.print(":");
  Serial.println(second(nowTime));
}

void loop(){}```

edgar-bonet, 2019-06-20:
I could not reproduce the problem on Linux with Arduino 1.8.5, using the current master branch of each of the libraries involved:

* RTClib: 5ddcf7c4d8aefa4286874426b158d7f3e0ec2b26 (Apr 13, 2019)
* Time: [d4fafb2][Time] (Jun 11, 2019)
* Timezone: [a913400][Timezone] (Nov 14, 2018)

Whether I include or not the line

```c++
DateTime ct;
```

I get the same result:

```text
1560889639 2019-6-18 20:27:19
```

[Time]: https://github.com/PaulStoffregen/Time/commit/d4fafb2c06c41a5ddf18f351bf30c5e22e1e7d2e
[Timezone]: https://github.com/JChristensen/Timezone/commit/a9134007b3c5a5274450474343bcabe06bc16292

lovethat777, 2019-06-20:
I used Arduino IDE 1.8.9 as well as 1.8.3 both on Windows (7 and XP) and deployed on a Pro Mini and a Nano, same results as mentioned. Also used the latest versions of RTClib, Time and Timezone (as you mentioned)

drak7, 2019-07-16:
**TL;DR**, weird optimization bug.

This is a weird one. Here's what happens as far as I can tell:

- RTClib.cpp defines daysInMonth as an array of bytes with the number of days in each month (31, 28, 31, 30, etc). This array is stored in PROGMEM.
- Time.cpp also does this with monthDays. The two arrays are exactly the same. This array is *not* stored in PROGMEM.
- The compiler sees two identical const arrays and helps us out by optimizing one away. It keeps the one in PROGMEM. Not sure if this is a bug or intentional.
- monthDays somehow still works as long as the DateTime constructor hasn't run.
- At run-time the default DateTime constructor calls pgm_read_byte. I don't know what happens here but something is modified and because monthDays is not accessed with pgm_read_byte it now gets the wrong data.

Workarounds:

- Add a dummy value to the end of one array, e.g. `static  const uint8_t monthDays[]={31,28,31,30,31,30,31,31,30,31,30,31,0};` The compiler now sees two different arrays and won't optimize it out. The code using monthDays knows there are only 12 months and doesn't mind the dummy value.
- Store monthDays in PROGMEM.
- Pretty much anything to fool the optimization, you could make monthDays local instead of global but then you're wasting some space. You can take daysInMonth out of PROGMEM and it'll work, they're optimized down to one array but now there's no PROGMEM access issue.

edgar-bonet, 2019-07-17:
We could get rid of the array and instead use arithmetic expressions to convert between long day counts and (year, month day) triplets. For example:

```diff
--- a/RTClib.cpp
+++ b/RTClib.cpp
@@ -109,9 +109,7 @@
 static uint16_t date2days(uint16_t y, uint8_t m, uint8_t d) {
     if (y >= 2000)
         y -= 2000;
-    uint16_t days = d;
-    for (uint8_t i = 1; i < m; ++i)
-        days += pgm_read_byte(daysInMonth + i - 1);
+    uint16_t days = d + (m * 153 - 151 - (m>2) * 11) / 5;
     if (m > 2 && y % 4 == 0)
         ++days;
     return days + 365 * y + (y + 3) / 4 - 1;
```

This also does away with the loop over the months, but it requires a division, which is not cheap on the AVR-based boards. It may end up being slower than the loop-based solution: some tests would be needed to tell for sure. This solution has also the drawback of not being obviously correct at first sight.

A similar method can be used to remove the only other use of `daysInMonth`, in `DateTime::DateTime(uint32_t t)`. That would remove two loops (over years and months) at the cost of three divisions (one in 32&nbsp;bits and two in 16&nbsp;bits).

Is this idea worth pursuing? I am not completely convinced, but if the maintainers think it is worth it, I could write it, test, and request a pull.

edgar-bonet, 2019-07-17:
Maybe a simpler solution would be to remove the December slot from the array. It looks like it is not very useful.

drak7, 2019-10-09:
Please open a PR @nobodyinperson and we'll merge this.

edgar-bonet, 2019-10-09:
As I wrote before, the issue can also be fixed by _removing_ rather than _adding_ an item. I just submitted a pull request that does so: #125.

commit 3069812e (2019-10-10): Remove the last cell in the daysInMonth array

This array is used in computations where the total number of days from
the previous months in the year is added or subtracted from a day count.
Since December is never among the previous months, the twelfth cell in
the array is not useful.

Fixes #114: an incompatibility with Paul Stoffregen's Time library
arising from both libraries defining identical arrays of constants.
closed 2019-10-10 by drak7