adafruit/RTClib #118: Lots of ugly "errors" in VSCode...

Daemach, 2019-07-30:
vscode does not like your syntax :/

https://www.screencast.com/t/CY0kzfrPLIE

edgar-bonet, 2019-07-30:
On the screenshot you provide, we see an error reported on [line 684 of RTClib.cpp][l684]:

```c++
uint32_t RTC_Micros::microsPerSecond = 1000000;
//                 ^^ ERROR HERE: “expected a ';'”
```

So the compiler is telling us it expected to see

```c++
uint32_t RTC_Micros;
```

which would only make sense _if_ `RTC_Micros` hasn't been defined before. Given that `RTC_Micros` has been defined as a class in RTClib.h, the expectation of the compiler does not make sense.

My guess is that your files got out of sync. You are compiling the latests RTClib.cpp with an older RTClib.h that does not define `RTC_Micros` (it's [a very recent addition][merge] to the library). Or maybe you have an older copy of the library lying around, your include paths got messed up, and the compiler is including the wrong RTClib.h.

This is not an issue with RTClib. You have to make sure you are not mixing up files from different versions of the library, and watch your include paths.

[l684]: ../blob/1.2.4/RTClib.cpp#L684
[merge]: ../commit/3f7d2e2b43528310264afb0b85602e322a600dce

Daemach, 2019-07-30:
I think vscode just lost its mind.  It's not showing errors this morning.  Sorry about the false start.

closed 2019-07-30 by Daemach