adafruit/RTClib #127: Check for invalid date in DateTime

ondras12345, 2019-11-24:
- Arduino board:  Arduino MEGA 2560
- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.9

The `DateTime::DateTime (uint16_t year, uint8_t month, uint8_t day, uint8_t hour, uint8_t min, uint8_t sec)` [constructor](https://github.com/adafruit/RTClib/blob/master/RTClib.cpp#L201) does not contain any user input checking, so it's possible to enter __invalid values__, such as month=13.

I can handle it in my code that uses this library, but it's harder to do so with incorrect day values, because I would have to check for leap. The library [already does that](https://github.com/adafruit/RTClib/blob/master/RTClib.cpp#L181) and I don't want to have to create a new array in my sketch as it would take up more space in the memory than reusing the [array](https://github.com/adafruit/RTClib/blob/master/RTClib.cpp#L119) that is already part of this library.

Setters in DateTime https://github.com/adafruit/RTClib/issues/70 (if someone implements them) would also need this checking.

It's bad to store incorrect date values, because some RTC don't check this either, eg. [DS3231 datasheet](https://datasheets.maximintegrated.com/en/ds/DS3231.pdf), page 12: 

> Illogical time and date entries result in undefined operation.

I may submit a pull request implementing both this and the setters, but I need to know what should happen if an invalid value is encoutered. Setters could just return false without changing anything, but I think the constructor would have to replace the value by a valid one because it can't just fail and do nothing.

wRorsjakz, 2020-01-13:
Hello. I am wondering if you managed to implement the error checking?

ondras12345, 2020-01-13:
> Hello. I am wondering if you managed to implement the error checking?

I haven't tried it yet, but it should be quite simple to do. 

Thanks for the reminder, I'm putting it on my TODO list.

wRorsjakz, 2020-01-13:
I've actually worked on it on my fork [here](https://github.com/wRorsjakz/RTClib/commit/4f279aca94a6567b8546b1a1679e5323f8d29236). I added setters for the attributes with error checking. But I realized there is also a need to have a [method](https://github.com/wRorsjakz/RTClib/blob/4f279aca94a6567b8546b1a1679e5323f8d29236/RTClib.cpp#L345-L361) to change the date (year, month, day) all at once instead of using just setters.

ondras12345, 2020-01-13:
Thanks.

I think the setters should also return false if the value is invalid.

Also, please don't use tabs to indent the code. The rest of the library uses spaces... (Well, except for the `char* DateTime::toString(char* buffer)` function)

You should also add the new keywords to the [keywords.txt file](https://github.com/wRorsjakz/RTClib/blob/master/keywords.txt).

ondras12345, 2020-01-14:
Hi,
you have removed a feature in [this commit](https://github.com/wRorsjakz/RTClib/commit/af382bfb7e7e25ff670ddcdd8da5a4d697be67ea). If the year is <2000, your version does not work while the documentation clearly states that 2 or 4 digits may be used to express the year.

Incorporate this into your code, please:
```
if (year >= 2000)
        year -= 2000;
```

`void setYear(uint16_t year)`, `bool DateTime::validDate(uint16_t year, uint8_t month, uint8_t day)` and `DateTime::DateTime (uint16_t year, uint8_t month, uint8_t day, uint8_t hour, uint8_t min, uint8_t sec)` need it.

Also, please explain the error checking in the documentation of `DateTime::DateTime (uint16_t year, uint8_t month, uint8_t day, uint8_t hour, uint8_t min, uint8_t sec)` - maybe something like:
```
If an invalid date is passed to the function, it gets replaced by 2000-01-01.
If an invalid value in any part of the time is encountered, it is replaced by 0.
```

edgar-bonet, 2020-04-16:
The constructor that takes year, month, day... as separate parameters does little more than store its arguments in the object. It is thus pretty easy to build an invalid `DateTime` with it. One can, however, check the validity of the constructed `DateTime` after the fact.

I just figured out a very simple way to check whether a `DateTime` is valid: convert to Unix time and back. You get the original `DateTime` if and only if it is valid.

My implementation:

```c++
bool DateTime::isValid() const {
    if (yOff >= 100) return false;
    DateTime other(unixtime());
    return yOff==other.yOff && m==other.m && d==other.d
           && hh==other.hh && mm==other.mm && ss==other.ss;
}
```

The initial test for `yOff >= 100` is there because only dates prior to 2100 are officially supported.

Is there interest in me submitting a pull request with this added method?

jadonmmiller, 2020-04-16:
> Is there interest in me submitting a pull request with this added method?

Absolutely! Personally, I don't understand why this hasn't been added before. Is there a reason it can't be added directly into that constructor? That seems like a cleaner solution to me.

Thanks for your suggestion, and I support any method of DateTime validation within the library!

ondras12345, 2020-04-16:
@wRorsjakz has a [fork](https://github.com/wRorsjakz/RTClib) with his version (and I have an [open PR](https://github.com/wRorsjakz/RTClib/pull/1) there), but he doesn't seem to respond.
Your solution looks much better, though.

edgar-bonet, 2020-04-16:
@jadonmmiller wrote:
> Absolutely!

OK, will do.

> Is there a reason it can't be added directly into that constructor?

Yes, a constructor cannot do something else that construct an object. The best it can do, in case it detects an error, is to return an invalid object. Turns out that's already what it does. ;-)

An `isValid()` method is all that's needed to formalize the concept of “invalid object”. This also seems to be [the approach taken by Qt][qt].

@ondras12345 wrote to @wRorsjakz:
> Please add something like: “_If an invalid date is passed to the function, it gets replaced by 2000-01-01. [...]_”.

I don't quite like the idea, as 2000-01-01 is a perfectly valid date.

[qt]: https://doc.qt.io/qt-5/qdatetime.html#isValid

jadonmmiller, 2020-04-16:
> The best it can do, in case it detects an error, is to return an invalid object. Turns out that's already what it does. ;-)

That was a dumb oversight on my part! I would've had it set the object to the earliest or latest date the system is capable of, but now that I think about it, that doesn't get us anywhere! 🤔 😄

commit ce96577b (2020-04-20): Add new method DateTime::isValid()

The documentation of `DateTime::DateTime(year, month...)' has been updated
to mention that a DateTime can be invalid and that its validity can be
tested with `isValid()'.

Fixes #127.
closed 2020-04-20 by drak7