adafruit/RTClib #131: Are Attiny85 still supported by RTCLib?

layzsmurf, 2020-01-22:
https://github.com/adafruit/RTClib lists Attiny85 as compatible. 

[Travis](https://travis-ci.com/adafruit/RTClib) however does not seem to check if the code compiles with an attiny85 environment - 
It seems to check the code for leonardo, zero, m4, esp32, esp8266, mega2560, uno .

- ~~Arduino~~ Attiny board:  Digispark

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.10
- RTCLib installed: 1.3.3

List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 
* Select a attiny board (Digispark (Default 16.5MHz)
* create an empty sketch and add the line on top '#include "RTClib.h"'
* save file
* compile

paste to error message: https://pastebin.com/E5bVMdiA

edgar-bonet, 2020-01-22:
AVR-based Arduino-compatible boards tend to have very limited RAM, and literal strings can be large RAM consumers. To mitigate this problem, the Arduino core provides the [`F()` macro][F], which is useful for keeping string literals in PROGMEM (i.e. flash) and thus saving RAM.

RTClib provides an overloaded [`DateTime` constructor][cstr] that takes a date and a time as PROGMEM strings. It relies on the undocumented `__FlashStringHelper` class, which is required since the `F()` macro returns a pointer to this class. The compiler error you see arises from the fact that the core you are using does not declare this class. It uses a [dummy `fstr_t` struct][fstr_t] instead.

Whose fault? It could be argued that it's RTClib's fault, as it is using an undocumented class. But note however that:

 * this class is absolutely needed for providing a PROGMEM string-based constructor
 * such a constructor is specially useful on an ATtiny, as these have still less RAM than your typical Arduino
 * the official Arduino cores provide `__FlashStringHelper` compatibility, [even on architectures where PROGMEM does not make sense][samd].

I would argue then that this issue should rather be filed against the [Digispark core][]. Actually, such an issue has [already been filed][issue-43], but it seems to have been ignored for the last three years. Funnily enough, their Digispark Pro core (which is also ATtiny-based) [does declare this class][pro core].

An workaround would be to add, near the top of RTClib.h, something along these lines:

```c++
#ifdef USING_DIGISPARK_CORE
# define __FlashStringHelper fstr_t
#endif
```

but I can't find a preprocessor macro that clearly identifies this particular core.

[F]: https://www.arduino.cc/reference/en/language/variables/utilities/progmem/#_the_code_f_code_macro
[cstr]: https://adafruit.github.io/RTClib/html/class_date_time.html#a75e22d94bd5ab883ec56e6581e04a2d7
[fstr_t]: https://github.com/digistump/DigistumpArduino/blob/1.6.7/digistump-avr/cores/tiny/Print.h#L54-L63
[samd]: https://github.com/arduino/ArduinoCore-samd/blob/v1.6.1/cores/arduino/WString.h#L37-L38
[Digispark core]: https://github.com/digistump/DigistumpArduino
[issue-43]: https://github.com/digistump/DigistumpArduino/issues/43
[pro core]: https://github.com/digistump/DigistumpArduino/blob/1.6.7/digistump-avr/cores/pro/WString.h#L37-L38

layzsmurf, 2020-01-22:
Thank you @edgar-bonet very much  for this in-depth description. I am really a beginner and was at a loss how to interpret the compiler error and where I could get started to fix it.

I added the following code to  in /Arduino/library/RTClib/RTClib.h and the compiler stopped complaining and after removing a short I got a running clock.
```
#if defined (__AVR_ATtiny85__)
# define __FlashStringHelper fstr_t
#endif
```
Which works, but might catch boards that did implement FlashStringHelper.

I have no idea what to do with this issue - it seems to be in no-mans land. Digispark did not fix it for 3 years and since it is an "unofficial" hack, they don´t have to. Same for RTClib, as attiny digispark boards might make up a neglectable number of use cases. What to do?

edgar-bonet, 2020-01-22:
I have no good solution for this problem. There are quite a few ATtiny Arduino-compatible cores floating around. I don't know how many of them support the ATtiny85. Your fix may clash with every one of those cores, save for the Digispark official one. It works for you, and that's fine, but it's probably not something to put in the official RTClib.

I suggest you try to patch the Digispark core instead. Maybe something like this (untested):

```patch
diff --git a/digistump-avr/cores/tiny/Print.cpp b/digistump-avr/cores/tiny/Print.cpp
index 7d303b7..b5781a1 100644
--- a/digistump-avr/cores/tiny/Print.cpp
+++ b/digistump-avr/cores/tiny/Print.cpp
@@ -101,7 +101,7 @@ void Print::print(double n, int digits)
   printFloat(n, digits);
 }
 
-int Print::print( fstr_t* s )
+int Print::print( __FlashStringHelper* s )
 {
   int rv;
   char ch;
@@ -179,7 +179,7 @@ void Print::println(double n, int digits)
   println();
 }
 
-int Print::println( fstr_t* s )
+int Print::println( __FlashStringHelper* s )
 {
   int rv;
 
diff --git a/digistump-avr/cores/tiny/Print.h b/digistump-avr/cores/tiny/Print.h
index 0e964b0..dea114d 100644
--- a/digistump-avr/cores/tiny/Print.h
+++ b/digistump-avr/cores/tiny/Print.h
@@ -49,19 +49,6 @@ public:
   virtual void print(Print &stream) const = 0;
 };
 
-/* ...Printable */
-    
-typedef struct
-{
-  char c;
-}
-fstr_t;
-
-/* rmv: Use the macro below in preparation for the next Arduino release.
-# define FSTR(s) ((fstr_t*)PSTR(s))
-*/
-# define F(s) ((fstr_t*)PSTR(s))
-
 class Print
 {
   private:
@@ -83,7 +70,7 @@ class Print
     void print(long, int = DEC);
     void print(unsigned long, int = DEC);
     void print(double, int = 2);
-    int  print( fstr_t* );
+    int  print( __FlashStringHelper* );
 
     void println(const String &s);
     void println(const char[]);
@@ -94,7 +81,7 @@ class Print
     void println(long, int = DEC);
     void println(unsigned long, int = DEC);
     void println(double, int = 2);
-    int  println( fstr_t* );
+    int  println( __FlashStringHelper* );
     int  println(void);
   public:
     /* Printable...*/
diff --git a/digistump-avr/cores/tiny/WString.h b/digistump-avr/cores/tiny/WString.h
index cadddb9..82b155b 100644
--- a/digistump-avr/cores/tiny/WString.h
+++ b/digistump-avr/cores/tiny/WString.h
@@ -25,6 +25,9 @@
 #include <string.h>
 #include <ctype.h>
 
+class __FlashStringHelper;
+#define F(string_literal) (reinterpret_cast<const __FlashStringHelper *>(PSTR(string_literal)))
+
 class String
 {
   public:
```

If it works for you, you may submit it to Digistump as a pull request. Sometime pull requests get more attention than issues.

caternuson, 2021-11-02:
This may work better now that the files have been broken up? And possibly other changes? Going to close, but can reopen if needed.

This complies OK for a target of "Adafruit Trinket"
```cpp
#include "RTClib.h"

void setup() {
}

void loop() {
}
```

closed 2021-11-02 by caternuson