adafruit/RTClib #138: Temperature function bug

Bit-Mac, 2020-03-25:
There is a bug in the RTClib.cpp  float RTC_DS3231::getTemperature() function.
The way the function is written does not recognize negative temperature values from the DS3231.  The 10-bit temperature is stored in 2's complement form.  An example of several ways to do this calculation are supplied below.  I used a simulator to run this code snipet.  It showed that my hunch was correct - that the RTClib.cpp code was not handling negative temperatures, while the other two methods worked just fine.  Try it  -- see what you think.  Bit-Mac.
//------------------------------------------------------------------------------
//	This is a DS3231 Temp math test program
//      DS3231 store a 2's complement, 10-bit Temp in two bytes

byte tMSB = -40;
byte tLSB = 0;
int sign = 1;
short iTemp = 0;
float fTemp = 0.0;
float fTemp2 = 0.0;
float fTemp3 = 0.0;

void setup()
{
	Serial.begin(9600);
	//First method uses 2's complement if neg Temp
	if (tMSB & 0x80) { sign= -1;}
	  else { sign = 1;}
	iTemp = (((short)tMSB << 8) | (short) tLSB);
	if (sign == -1) {
		iTemp = (iTemp ^ 0xFFFF) + 1;
	}
	fTemp = sign * (iTemp/256.0);
	Serial.println (fTemp,2);
	Serial.println (tMSB);
	Serial.println (tLSB);
	
	//Second method is from DS3231-master and RTClib
	fTemp2 = ((((short) tMSB << 8) | (short) tLSB) >> 6)/4.0;
	Serial.println (fTemp2,2);
	
	//Third method is from DS3231RTClib - Adafruit RTClib
	//This method ignores negative Temps
	fTemp3 = (float) tMSB + (tLSB >>6) * 0.25;
	Serial.println (fTemp3, 2);
	
	
}

void loop()
{
	
}
//-------------------------------------------------------------------

edgar-bonet, 2020-03-25:
The first method seems to have a special handling for negative numbers, in order to correctly convert them to the internal processor format. This special handling is useless, as the internal processor format is two's complement. The statement

```c++
iTemp = (((short)tMSB << 8) | (short) tLSB);
```

already gives the correct signed value in `iTemp`. The second method relies on this.

BTW, [left-shifting a negative number is undefined behavior in C++][UB]. It would be safer to shift the MSB as an unsigned number, then casting the result to a signed number. This extra cast costs zero machine instructions:

```c++
float fTemp = ((int16_t) (((uint16_t) tMSB << 8) | tLSB) >> 6) * 0.25;
```

or maybe even

```c++
float fTemp = (int16_t) (((uint16_t) tMSB << 8) | tLSB) * (1/256.0);
```

if we do know that the last 6 bits are zeros (I did not check on the datasheet).

[UB]: https://stackoverflow.com/questions/9867883/is-left-shifting-a-signed-integer-undefined-behavior-in-c03

Bit-Mac, 2020-03-25:
[RTClib Test Temp Results.txt](https://github.com/adafruit/RTClib/files/4383236/RTClib.Test.Temp.Results.txt)
Thanks for your instructive comments.  I have attached some further comments and a real example for the problem and possible solution. Thanks. BitMac

edgar-bonet, 2020-03-25:
@Bit-Mac wrote:

```c++
float RTC_DS3231::getTemp1()
{
  uint8_t lsb; // Made var "msb" a signed 8-bit type which allows float to retain sign
  char msb; // NOTE: char, but also +127/-128 signed 8-bit value
  // rest is unchanged relative to RTC_DS3231::getTemperature()
}
```

This is a nice solution, as it solves the problem while minimally changing the code. However, the C++ language does dot specify whether `char` is signed or not. Gcc happens to use signed chars by default but, for the sake of clarity, I would recommend being more explicit and declaring `msb` as an `int8_t`.

@Bit-Mac, would you submit a pull request? I can do it if you prefer.

Bit-Mac, 2020-03-25:
Edgar,
Thanks for the informative comment.  I'm still learning  arduino c++ and had not seen a 8-bit signed int type (int8_t) as yet. Learning something new every day.  I am not sure what a pull request is since I just signed on to Git-Hub last night. I think you must be talking about a change request of some kind? How do I do that? I am willing to try, if you could give me some hints. Thanks, Bit-Mac

edgar-bonet, 2020-03-26:
> I [...] had not seen a 8-bit signed int type (int8\_t) as yet.

With gcc, it's actually equivalent to `char`. The differences are:

* `int8_t` is guaranteed to be signed on every platform
* `int8_t` more clearly conveys that the programmer's intent is to have an 8-bit signed integer, whereas `char` connotes (and is often used for storing) a character.

> I am not sure what a pull request is since I just signed on to Git-Hub last night. I think you must be talking about a change request of some kind?

Exactly. See [About pull requests][about-pr] on the GitHub help pages.

> How do I do that?

The quickest way would be to click on the pencil icon near the top right of the page [RTClib.cpp][] (just above the code). The tooltip says “Fork this project and edit the file”. Once edited, you will be given the option to write a short message describing the change and submit a pull request. The problem with this method is that you cannot test the change locally before submitting the pull request.

The canonical method is to
* fork the repository on GitHub
* clone the fork on your own computer
* create a new branch for your change
* edit the file on the working copy of your local clone
* test
* commit the change, with a short descriptive message, e.g. “DS3231: support reading negative temperatures”
* push the branch to your fork
* submit the pull request

This is more involved, as it requires some knowledge of git. On the other hand it lets you do most of the work (including testing) on your own computer. See [Forking Projects][forking] for a quick guide, or search Google or YouTube for “pull request tutorial”.

[about-pr]: https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/about-pull-requests
[RTClib.cpp]: https://github.com/adafruit/RTClib/blob/master/RTClib.cpp
[forking]: https://guides.github.com/activities/forking/

drak7, 2020-03-27:
Merged #139 ! Thanks for finding this.

closed 2020-03-27 by drak7