adafruit/RTClib #143: Strange Serial prints when using DateTime library

jadonmmiller, 2020-04-09:
The sketch below should simply echo whatever is sent over the serial connection. However, if you create a DateTime object before the Serial.print it echos bogus data. It looks like it might be a memory overflow or something? In the code where I originally noticed the problem, it would print random segments of previous prints, variable values, etc. (Weird, I know!) To make the code work as intended, comment out the lines relating to the DateTime object.

I tried using a different constructor as well, but that didn't help.

```C++
#include <RTClib.h>

//Accumulates the incoming serial data
char incoming[101] = {0};

void setup() {
  Serial.begin(9600);

  //Comment out these lines to make the code function properly
  //--------------------------------------------------------------
  DateTime compiledOn(F(__DATE__), F(__TIME__));
  char buffer[] = "MMM DD, YYYY hh:mm";
  Serial.println(compiledOn.toString(buffer));
  //--------------------------------------------------------------
}

void loop() {
  if (Serial.available()) {

    char newByte = Serial.read();

    if (newByte == '\n') { //termination character
      Serial.println(incoming); //print the accumulated data

      //Clear the string
      for (uint8_t i = 0; i < 100; i++) {
        incoming[i] = 0;
      }
    }
    else { //newByte not '\n'
      strcat(incoming, &newByte); //accumulate the incoming data
    }
  }
}
```

- Arduino board:  **Arduino Uno**

- Arduino IDE version: **1.8.12**

- RTClib version: **1.4.1**

- Steps to reproduce:
1. Upload above sketch to Arduino
2. Send anything to the Arduino over Serial
3. Arduino should echo what you sent, but will reply with bogus data
4. Comment out the indicated lines and upload again
5. Sketch should now echo data properly

Please feel free to suggest any ideas you have - I'm stumped!

edgar-bonet, 2020-04-09:
This is wrong:

```c++
strcat(incoming, &newByte);
```

`strcat()` expects both arguments to be NUL-terminated strings, which is not the case of `&newByte`. You have what is known in C++ as “undefined behavior”, which means that _anything_ can happen, including the kind of behavior you are experiencing.

This issue should be closed as invalid. This issue tracker is intended for reporting bugs in the library. The [Arduino forum](https://forum.arduino.cc/) and [arduino.stackexchange](https://arduino.stackexchange.com/) would be better places for discussing a problem like this one.

jadonmmiller, 2020-04-09:
> `strcat()` expects both arguments to be NUL-terminated strings, which is not the case of `&newByte`. You have what is known in C++ as “undefined behavior”, which means that _anything_ can happen, including the kind of behavior you are experiencing.

Yes, you're right. I'm sorry for wasting your time. I don't know how I overlooked that!

> This issue should be closed as invalid. This issue tracker is intended for reporting bugs in the library. The [Arduino forum](https://forum.arduino.cc/) and [arduino.stackexchange](https://arduino.stackexchange.com/) would be better places for discussing a problem like this one.

My bad. I thought I had enough evidence to indicate a library issue. Sorry again, and thanks for your help.

closed 2020-04-09 by jadonmmiller