adafruit/RTClib #144: Leap year and daylight saving time

Eldenroot, 2020-04-12:
Hi, is possible somehow automatically detect DST and leap year in this library? If yes, can you please add an example?

Thank you!

edgar-bonet, 2020-04-12:
DST: it is not possible to automatically detect. The rules for switching from/to DST vary by country. In some countries (e.g. the US) they can even vary by region. Not even that, but they are changed occasionally by governments. It boils down to political decisions that are completely unpredictable. On a desktop OS, you typically have a large database of DST rules applicable to each timezone, and the database is periodically updated by the OS vendor.

Leap years: these are already handled by the RTC hardware, and the library is also aware of them. If you build a `DateTime` for 28&nbsp;February and add to it a `TimeSpan` of 1&nbsp;day, you get either 29&nbsp;February or 1<sup>st</sup>&nbsp;March depending on the year.

Eldenroot, 2020-04-12:
ok, what about this PR? 

https://github.com/adafruit/RTClib/pull/36/files

edgar-bonet, 2020-04-12:
> what about this PR?

The DST rules used across the EU have been consistent for quite a while, although there is now significant political pressure for breaking this consistency. This PR specifically implements the current EU DST rules, so it could help you since you seem to live en Europe. However, they way the code is written, it assumes the winter UTC offset is +00:00, which is only the case in the UK, Ireland, Portugal, and some islands in the Atlantic. You could easily patch the code for the CZ timezone though.

> And any idea how to easily add DST detection into my code?

The simplest option would probably be to use this PR, and replace the test for `dt.summertime_EU(0)` by `dt.summertime_EU(1)`. Note that I have no idea whether the implementation of the PR is good.

Eldenroot, 2020-04-12:
OK, thank you.

I know that there is a pressure to not use DST... but I think it will not be soon (mainly because there are other issues right now - CO2 level in air, air pollution and COVID-19).

Maybe would be nice to reconsider and add a support into library itself with configurable UTC offset and conditions. Thank you anyway!

drak7, 2020-04-14:
There are so many different rules, conditions, etc for DST in different countries that I think it'd be better to leave up to the user or have a separate library just for that.

closed 2020-04-14 by drak7