adafruit/RTClib #145: Revision history

tqueale, 2020-04-14:
A clue on what has changed might be a good idea.

edgar-bonet, 2020-04-14:
A short (too short maybe?) description of the changes is available in [the Releases page](../releases). The full details can be extracted from the git log, e.g.:

```shell
git log --first-parent --date=short --format="%ad%d %s"
```

I could try to build a `Changes.md` from the output, that would look like this:

* [1.4.1](../releases/tag/1.4.1) — 2020-04-06
  * Let `getTemperature()` handle negative temperatures (#139)
* [1.4.0](../releases/tag/1.4.0) — 2020-03-10
  * Add support for DS3231 alarms (#135)
* [1.3.4](../releases/tag/1.3.4) — 2020-02-28
  * ds3231.ino: add commented-out `rtc.adjust()` (#134)
  * ds1307.ino: add commented-out `rtc.adjust()` (#133)
* [1.3.3](../releases/tag/1.3.3) — 2019-10-24
  * Test I2C connection in `RTC_DS3231::begin()` (#126)
* ...

However, keeping the file up to date ads some maintenance burden. Creating the file would only be worthwhile if the maintainers are willing to invest the time to keep it up to date.

drak7, 2020-04-14:
I think Git takes care of this pretty well.

closed 2020-04-14 by drak7