adafruit/RTClib #146: What is the valid range for DateTime?

edgar-bonet, 2020-04-15:
I am planning to prepare a pull request that expands the [documentation of the `DateTime` class][DateTime], which I find somewhat terse and incomplete as it stands now. I would like to add a clear statement about the range of dates that the class can correctly handle. It seems quite clear to me that the earliest valid date is 2000-01-01, but what is the other end of the range?

Looking at the source code, there seem to be various dates at which different parts of the implementation break:

| date                | breakage                          |
|---------------------|-----------------------------------|
| 2068-01-19 03:14:08 | return type of `secondstime()`    |
| 2100-01-01 00:00:00 | [comment for `date2days()`][c]    |
| 2100-03-01 00:00:00 | computation of leap years         |
| 2106-02-07 06:28:16 | return type of `unixtime()`       |
| 2136-02-07 06:28:16 | `secondstime()`, if made unsigned |
| 2179-06-07 00:00:00 | return type of `date2days()`      |
| 2256-01-01 00:00:00 | data type of `yOff`               |

 * The first limit is probably an oversight that can be easily fixed by returning `uint32_t` instead of `long`. This would raise the limit of `secondstime()` to 2136-02-07.
 * The second is just a comment, though I understand the value of specifying a limit that has no technical justification but is easy to remember.
 * The computation of leap years can be easily fixed, but that would add a small cost in code size and execution time. It may not be worth it.
 * The `unixtime()` limit would be harder to fix without resorting to 64-bit arithmetics. One option would be to specify that the method returns only the lowest 32&nbsp;bits of the whole result, and let the user figure out the other bits if she needs them, following the approach of NTPv3. Calculations that use `unixtime()` internally would switch to using `secondstime()`.
 * Beating the 2136 limit would probably be way too costly.

I am leaning towards specifying the limit as either 2100-03-01 (that date excluded, to avoid the cost of a better leap year calculation) or 2100-01-01 (nice round number easy to remember). But since I am in no position to decide, I need feedback from the maintainers.

[DateTime]: https://adafruit.github.io/RTClib/html/class_date_time.html
[c]: ../blob/1.4.1/RTClib.cpp#L123

drak7, 2020-04-15:
I like 2100-01-01 for an "official" limit. If this is still running in 80 years, great!

edgar-bonet, 2020-04-15:
@drak7: Thanks for your feedback! I do like the option. It may be the one that makes most sense given that the supported RTCs store the year as two decimal digits in BDC.

closed 2020-04-15 by edgar-bonet