adafruit/RTClib #152: ESP8266 problem

Lightwell-bg, 2020-04-22:
RTClib@1.4.2 version doesn't work with ESP8266

fatal error: avr/interrupt.h: No such file or directory

edgar-bonet, 2020-04-22:
Please, provide the detailed steps to reproduce the error.

There are nine example sketches bundled with the library. All of them have been [successfully compiled for ESP8266][esp] by the automated [test script][build].

[esp]: https://github.com/adafruit/RTClib/runs/605549893?check_suite_focus=true#step:6:86
[build]: https://github.com/adafruit/ci-arduino/blob/master/build_platform.py

Lightwell-bg, 2020-04-22:
Before update to 1.4.2 all was ok.
After I cant compile sketch (fatal error: avr/interrupt.h: No such file or directory in USI_TWI_Master.cpp)  .
I use PLATFORMIO.

edgar-bonet, 2020-04-22:
Try removing the last line from RTClib/library.properties:

```diff
diff --git a/library.properties b/library.properties
index 44a50c4..97b70be 100644
--- a/library.properties
+++ b/library.properties
@@ -7,4 +7,3 @@ paragraph=A fork of Jeelab's fantastic RTC library
 category=Timing
 url=https://github.com/adafruit/RTClib
 architectures=*
-depends=TinyWireM, OneWire
```

Alternatively, try [this version][rm-depends]: it's just 1.4.2 that line removed.

Tell us if it fixes the issue for you. If that's the case, I'll submit a pull request.

[rm-depends]: https://github.com/edgar-bonet/RTClib/archive/rm-depends.zip

ladyada, 2020-04-22:
@ivankravets hihi - taggin you to see another example that platform-unique dependencies cause compilation issues. tinywirem is referenced here:
https://github.com/adafruit/RTClib/blob/a40bfbc148f24eceabfe5712b03ba430ca0649ac/RTClib.cpp#L45

ivankravets, 2020-04-22:
Hi @ladyada, thanks for tagging me. Yes, yes, this is a known issue. There are a few solutions:
1. Using `library.json` and set `build.libLDFMode` to `chain+`. However, I remember that you would like to have only `library.properties` in a repository. See solution 2
2. Update LDF mode per PlatformIO project as described here https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-ldf-mode

We have some ideas about how to improve our Library Dependency Finder but it will need some efforts. The current LDF behavior is to follow your declarations. It allows PlatformIO Core to build project crazy fast.

Lightwell-bg, 2020-04-22:
> Try removing the last line from RTClib/library.properties:
> 
> ```diff
> diff --git a/library.properties b/library.properties
> index 44a50c4..97b70be 100644
> --- a/library.properties
> +++ b/library.properties
> @@ -7,4 +7,3 @@ paragraph=A fork of Jeelab's fantastic RTC library
>  category=Timing
>  url=https://github.com/adafruit/RTClib
>  architectures=*
> -depends=TinyWireM, OneWire
> ```
> 
> Alternatively, try [this version](https://github.com/edgar-bonet/RTClib/archive/rm-depends.zip): it's just 1.4.2 that line removed.
> 
> Tell us if it fixes the issue for you. If that's the case, I'll submit a pull request.

It doesnt work

ivankravets, 2020-04-22:
@Lightwell-bg does it work if you add to platformio.ini
```
[env]
lib_ldf_mode = chain+
```

Lightwell-bg, 2020-04-22:
> lib_ldf_mode = chain+

Sorry, and it doesnt work
                                                 ^
.pio\libdeps\d1_mini\TinyWireM_ID797\USI_TWI_Master.cpp:22:27: fatal error: avr/interrupt.h: No such file or directory
 #include <avr/interrupt.h>

ivankravets, 2020-04-22:
@valeros, please help

valeros, 2020-04-22:
Hi @Lightwell-bg ! The error is caused by a misleading dependency specified in `library.properties`  (`TinyWireM` is not supposed to work with `esp8266`). The best solution is to add the following lines to your `platformio.ini`:
```
[env:myenv]
framework = arduino
platform = espressif8266
board = ...

# explicitly ignore TinyWireM
lib_ignore =
    TinyWireM
# Resolve hidden dependencies under ifdef
lib_ldf_mode = deep+
```

Lightwell-bg, 2020-04-23:
Hi @valeros 
Thank you, it's work.

closed 2020-04-23 by ladyada

rgl, 2020-07-19:
Isn't there a better way to solve this issue?

Using lib_ignore in our platform.ini file feels dirty.

ivankravets, 2020-07-19:
Hey @ladyada ,

Would you like to have `library.json` which resolves these conflicts? We will explain in manifest when to use particular dependencies. You don't need to maintain it later, just increment "version" field (the same what you do for library.properties)

ladyada, 2020-07-20:
hihi @ivankravets  - as other library developers such as @paulstoffregen have mentioned, folks usually forget or neglect to update both on the libs they maintain. https://github.com/adafruit/DHT-sensor-library/pull/118#issuecomment-457876625

as more libraries add dependencies, some of which are for specific platforms, we recommend only compiling code that is referenced/included in the projects.

rgl, 2020-07-20:
@ivankravets, If I understood correctly, we cannot express a conditional library dependency using `library.properties` but we can in a `library.json` file? If so, can you provide an example of this?

Maybe we can automatically generate a `library.properties` from the `library.json` file? that way we only need to maintain one of them?

PaulStoffregen, 2020-07-20:
@ivankravets - Please consider the long term impact of solving problems by suggesting edits to a global config file or installing a specially modified copy of a library to solve a specific problem.

One of the commonly reoccurring tech support problems we see with the Arduino IDE is a user installed a modified copy of a library months or years ago, to fix some specific problem they were having at the time.  Often they were just following a suggestion someone gave them on a forum, without any idea of *why* installing a modified copy fixed their immediate issue.  They're just happy to be able to move past a compile error and get their project working.

Then months or years later, they buy a different board for a new project.  The user has long since forgotten about the prior problem.  Even at the time, they didn't really understand they were installing a custom copy or making a global chance.... they just followed advice from a forum message and it solved their problem.  But later it causes new problems when they use a different board.  Those new problems are **MUCH HARDER** for anyone on forums to help diagnose, because now the user is running with a modified platform or customized copies of libraries, which they do not even know.  How can those of us who answer tech questions on forums solve problems for these users.  We don't know they have modified files.  They don't know they have modified files.

Please, I beg of you, consider the long-term consequence of giving this sort of help where you direct people to modify their config files or install edited copies of libraries.  Those changes will persist on their computer and later are likely cause much harder tech support problems.

Please consider this tech support scenario in the design of your software.  Years ago I contributed code to the Arduino IDE which prints the full pathnames when 2+ libraries have the same header and the IDE had to choose 1 of them.  That code came as a result of answering many of these difficult tech support questions, where we had to guess that the user didn't have the original libraries installed.

PaulStoffregen, 2020-07-20:
As you design PlatformIO, please keep in mind your "users" are not only the people running your software, but also those of us who they will ask for help when things go wrong.

For another example of code I contributed to Arduino, when code fails to compile in Arduino this "Copy error messages" button appears.

[image]

This Arduino contribution was inspired by a 60+ message thread where we repeated begged a user to show us the Arduino IDE's error messages.  This button is meant to give users a gentle suggestion to copy the actual error messages when they ask for help on forums.  Since this button went into Arduino's software, we've had much high success solving user problems on forums, because some of them use it to put the errors into their question, and others who don't get the visual hint are more able to do so when we ask them to show the errors.

I'm taking the time to write these messages because in recent months I am starting to see many more people using PlatformIO instead of Arduino.  But I'm also not so happy trying to help those users.  When they don't show the errors, when their software might be set up in different ways, helping people is much harder.

Please, think of this as you continue to develop PlatformIO.  I and many others who answer forum questions are not direct users, but we do try to help everyone solve technical problems.  The more you do in PlatformIO to encourage users to better ask questions on forums, and the more consistent the software setup is (or more visually apparent any customizations are when errors happen) the better we can help your users to get their projects working.

ivankravets, 2020-07-21:
I've just fixed this issue https://github.com/platformio/platformio-core/issues/3607

@rgl , Lightwell-bg , and others. Could you upgrade PIO Core to the latest development version via `pio upgrade --dev` command and remove `lib_ignore` option from `platformio.ini` file. Does it work now?

ivankravets, 2020-07-21:
> But I'm also not so happy trying to help those users. 

So, you are happy to help Arduino's users who do not use your hardware and do not happy to help your (not ours) customers who bought your hardware and program it with PlatformIO which is free and open source. What is the reason?

Every day thousands of developers use PlatformIO to program their Teensy boards. Why you don't count these happy users? The rate of users who had some issues is very very small. Next time please forward them to https://community.platformio.org/ and our awesome community will help.

> in recent months I am starting to see many more people using PlatformIO instead of Arduino. 

We don't do any promotions and would be happy if you put information about PlatformIO on your website and give people a chance to have a choice.

There is 1,000,000 unique installations **just for VSCode+PlatformIO** (we support 10+ other IDEs/editors):
- https://marketplace.visualstudio.com/items?itemName=platformio.platformio-ide

Also, there are the answers why they move away from Arduino and other vendor IDEs (everyday fresh and successfully stories):
- https://marketplace.visualstudio.com/items?itemName=platformio.platformio-ide&ssr=false#review-details

---------------

PlatformIO is **THE ONLY** neutral and independent ecosystem for embedded development today. We encourage everyone to join our small revolution and give true freedom for embedded developers and businesses.

P.S: The more resistant is to PlatformIO, the more people use it (law of nature).

PaulStoffregen, 2020-07-23:
> What is the reason?

Maybe you didn't understand?  When we get PlatformIO users with problems, usually they give us almost zero info.  Sometimes they do not even say they are using PlatformIO.

Worst of all, the *way* you are advising your users to fix problems in this thread and others I have seen is with configuration changes.  Did you not read any of the words I wrote about how these sorts of changes (hacked copies of libraries, global config changes) persist on people's computers and later cause trouble?

Lack of info from users who ask for help and so much uncertainty about what they really have installed on their computer does not make me feel good about helping PlatformIO users.

ivankravets, 2020-07-26:
> Maybe you didn't understand? When we get PlatformIO users with problems, usually they give us almost zero info. Sometimes they do not even say they are using PlatformIO.

What do you propose now? To kill me because users do not report details? If there is an issue with a build process, they see the reason in the console. We don't hide errors from them. You try to build your picture about PlatformIO on some users who contacted you and didn't provide "information" which you wanted to see "in the first post". Please look widely...

> fix problems in this thread and others I have seen is with configuration changes.

Do you mean this is bad that PlatformIO allows developers to have full control over the build process where they can pass their own build flags, remove existing, control dependencies, and do other professional things? This is how PlatformIO differs from Arduino. There is no INO (which makes trouble for people who would like to do native C/C++ programming) or 1 sketch - there is a project. Developers use PlatformIO for PROJECTS, not for the 1-file sketches.

We don't ask people to change the configuration of a LIBRARY or LIBRARY SOURCE code. We ask them to configure PlatformIO LDF behavior in their `platformio.ini` project configuration file. Please re-read whole my posts. You were not so attentive. The discussion was about this https://docs.platformio.org/en/latest/projectconf/index.html

Please take a look at what people do with PlatformIO and after that, all questions about "why they don't use Arduino" will be closed. They technically can't use Arduino due to its concept and 1-sketch limitation. It's good for children or for beginners. This is why we also recommend them Arduino. PlatformIO is not a replacement for the Arduino, it's a continuation of your work if you can't live with "notepad" and 1 sketch where you don't have any control.
- https://github.com/MarlinFirmware/Marlin/blob/2.0.x/platformio.ini
- https://github.com/esphome/esphome/blob/dev/platformio.ini
- https://github.com/xoseperez/espurna/blob/dev/code/platformio.ini

> Did you not read any of the words I wrote about how these sorts of changes (hacked copies of libraries, global config changes) persist on people's computers and later cause trouble?

I saw that. This is not a truthful accusation. It's very bad that people do not remember the past. The same recently did your friend @mbanzi. You can read this thread https://twitter.com/mbanzi/status/1257980326808338434

**What do I mean here?**

1. "hacked copies of libraries" - this was Arduino workflow. They forced people to use the ZIP archive of libraries. 
2. Arduino forced people to manage libraries on a machine as ZIP archive where they modify them directly
3. Arduino gave them only ONE way to install libraries - from sources, ZIP archives, or via manual copying.
4. As a result, we had a ton of the same cloned, modified, and copied libraries on the internet.

# Library Management Solution

PlatformIO was THE FIRST project in an embedded world which proposed dependency management solutions based on Semantic Versioning and using packages from the REGISTRY. That was the reason why we created the first registry for the embedded community. 

It was 2014 and it's very bad when people don't remember history. What is worse, YOU were the part of that discussion where we wanted to improve Arduino IDE as well. If you don't remember, Google remembers that:

- https://groups.google.com/a/arduino.cc/g/developers/c/xCDjekcQW4k/discussion?pli=1
- https://github.com/PaulStoffregen/OctoWS2811/pull/2

Again, it was 2014. Later, we will see some attempts from Arduino.cc to replicate the same as what we did in 2017. The same we will see with Arduino CLI which replicates PlatformIO in 2019, and now we see Arduino IDE Pro which replicates PlatformIO for VSCode. The only problem here that there is a gap with 2-3 years for this replication...

We are so proud that other companies follow our vision and we are glad to help them with PlatofmrIO adoption as much as we can. The community will benefit from that.

# Conclusion

1. I don't want to discuss these accusations without arguments. 
2. If you want to have answers from me, please provide links/examples where I can get details about what is going wrong with  users who contact you
3. Please do not treat PlatformIO as "Ivan". There is the PlatformIO Team and other people who work on PlatformIO and you need to respect it. If you have something personal, it's better to contact me@ikravets.com

Finally, we love the critic when it is "sober" and has arguments. It helps us to make PlatformIO better and better. You can't make a product where 100% of people will be satisfied with it. The same with PlatformIO. We have our own philosophy, unique vision and there is an army from developers who love so much what we do. 

Arduino is a VERY VERY small puzzle in our ecosystem. We support 20+ other frameworks/SDKs/RTOS, such as FreeRTOS, Zephyr RTOS, Arm Mbed, STM32Cube, etc. We support the Arduino framework to help people with migration into the professional instruments which I mentioned before when their products go into production. All of us understand that Arduino != Professional/Production. This is where we play a significant role and reduce barriers when project requirements change and you need more advanced tools. PlatformIO saves time to market costs here.

Have a great weekend!

drak7, 2020-07-26:
This has moved well beyond the scope of the original issue and is not the place to argue about the merits of one tool vs another. Please take it to a more appropriate forum.

tannewt, 2020-07-27:
FYI, I've hidden the heated conversation as off-topic. It's not helpful.