adafruit/RTClib #161: DST settings

Eldenroot, 2020-04-28:
DST settings support :)

edgar-bonet, 2020-04-28:
Isn't this a duplicate of #144?

What kind of support do you expect? Switching DST on and off is trivial to do at the user level by simply adjusting the time by ±1h:

```c++
static inline void startDST() {
    rtc.adjust(rtc.now() + TimeSpan(3600));
}

static inline void stopDST() {
    rtc.adjust(rtc.now() - TimeSpan(3600));
}
```

It is then up to you to provide an appropriate UI for the end user to enable/disable DST. You may want to save the current DST status in NVRAM or EEPROM in order to show it on that UI.

Automatic handling of DST is trickier. If this is what you want, I suggest you keep the RTC in UTC, and convert to local time when needed:

```c++
DateTime localTime = utcToLocal(rtc.now());
```

This is how DST is handled in Unixish systems. You will have to implement `utcToLocal()` yourself: as it has already been explained to you, there are too many different rules, and too frequent changes to these rules, for this to be supported by the library.

If you only care about the current EU rules, you may adapt [this function][eu_dst] I once submitted to the avr-libc maintainers:

```c
int eu_dst(const time_t * timer, int32_t * z)
{
    uint32_t t = *timer;
    if ((uint8_t)(t >> 24) >= 194) t -= 3029443200U;
    t = (t + 655513200) / 604800 * 28;
    if ((uint16_t)(t % 1461) < 856) return 3600;
    else return 0;
} 
```

Keep in mind that avr-libc uses the 2000.0 epoch. You will have to adapt that to the 1970.0 epoch used by `DateTime::unixtime()`. There is already a macro in the library for the offset: `SECONDS_FROM_1970_TO_2000`.

[eu_dst]: http://savannah.nongnu.org/bugs/?44327

drak7, 2020-04-28:
Duplicate of #144, this is better left to the user or a separate library.

closed 2020-04-28 by drak7