adafruit/RTClib #171: Compability - PCF85063A

allexotecnologia, 2020-06-17:
- Arduino board:  **ESP32 DEVKIT**

Hey. 

Been looking for a RTC module and found out the NPX PCF85063A. Found this lib that works with PCF8563.

Any chance those are compatible using this same library?

Thanks

fm-33, 2020-06-17:
Hi!
If you already bought the module, test it and tell us! Many modules use the same methods.
If you can use another module, try DS3231. It is very reliable and widely used (and also cheap).

drak7, 2020-06-17:
The registers on the PCF85063A are different from the ones in the PCF8523 so the library won't work without modifications.

allexotecnologia, 2020-06-19:
> The registers on the PCF85063A are different from the ones in the PCF8523 so the library won't work without modifications.

@drak7 thanks for the reply. On that topic, any idea if that’s too complex to develop? Or how do I even begin? We had a spec for a project that uses this rtc and wishes to use it with a i2c bus linked on a esp32.

Thanks in advance!

edgar-bonet, 2020-06-19:
@allexotecnologia: You have to read the datasheet, and pay special attention to the description of the internal registers. Then you can take inspiration from the implementation of the existing `RTC_*` classes.