adafruit/RTClib #175: rtc libary in visual studio code

jeroen24, 2020-07-09:
hey,
i am trying to youse the exampel  code . but every time i get this error .
i think dear is a libary to short in visual studio code but i dont now witch.( i already installed TinywireM).

can enyone help me?
this is  the errors i get:
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp: In function 'void USI_TWI_Master_Initialise()':
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:64:3: error: 'PORT_USI' was not declared in this scope
   PORT_USI |=
   ^
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:66:11: error: 'PIN_USI_SDA' was not declared in this scope
        << PIN_USI_SDA); // Enable pullup on SDA, to set high as released state.
           ^
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:69:11: error: 'PIN_USI_SCL' was not declared in this scope
        << PIN_USI_SCL); // Enable pullup on SCL, to set high as released state.
           ^
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:71:3: error: 'DDR_USI' was not declared in this scope
   DDR_USI |= (1 << PIN_USI_SCL); // Enable SCL as output.
   ^
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:74:3: error: 'USIDR' was not declared in this scope
   USIDR = 0xFF; // Preload dataregister with "released level" data.
   ^
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:75:3: error: 'USICR' was not declared in this scope
   USICR = (0 << USISIE) | (0 << USIOIE) | // Disable Interrupts.
   ^
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:75:17: error: 'USISIE' was not declared in this scope
   USICR = (0 << USISIE) | (0 << USIOIE) | // Disable Interrupts.
                 ^
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:75:33: error: 'USIOIE' was not declared in this scope
   USICR = (0 << USISIE) | (0 << USIOIE) | // Disable Interrupts.
                                 ^
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:76:17: error: 'USIWM1' was not declared in this scope
           (1 << USIWM1) | (0 << USIWM0) | // Set USI in Two-wire mode.
                 ^
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:76:33: error: 'USIWM0' was not declared in this scope
           (1 << USIWM1) | (0 << USIWM0) | // Set USI in Two-wire mode.
                                 ^
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:77:17: error: 'USICS1' was not declared in this scope
           (1 << USICS1) | (0 << USICS0) |
                 ^
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:77:33: error: 'USICS0' was not declared in this scope
           (1 << USICS1) | (0 << USICS0) |
                                 ^
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:78:17: error: 'USICLK' was not declared in this scope
           (1 << USICLK) | // Software stobe as counter clock source
                 ^
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:79:17: error: 'USITC' was not declared in this scope
           (0 << USITC);
                 ^
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:80:3: error: 'USISR' was not declared in this scope
   USISR = (1 << USISIF) | (1 << USIOIF) | (1 << USIPF) |
   ^
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:80:17: error: 'USISIF' was not declared in this scope
   USISR = (1 << USISIF) | (1 << USIOIF) | (1 << USIPF) |
                 ^
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:80:33: error: 'USIOIF' was not declared in this scope
   USISR = (1 << USISIF) | (1 << USIOIF) | (1 << USIPF) |
                                 ^
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:80:49: error: 'USIPF' was not declared in this scope
   USISR = (1 << USISIF) | (1 << USIOIF) | (1 << USIPF) |
                                                 ^
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:81:17: error: 'USIDC' was not declared in this scope
           (1 << USIDC) |    // Clear flags,
                 ^
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:82:19: error: 'USICNT0' was not declared in this scope
           (0x0 << USICNT0); // and reset counter.
                   ^
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp: In function 'unsigned char USI_TWI_Start_Transceiver_With_Data(unsigned char*, unsigned char)':
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:161:13: error: 'USISIF' was not declared in this scope
       (1 << USISIF) | (1 << USIOIF) | (1 << USIPF) |
[... 204 lines cut ...]
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp: In function 'unsigned char USI_TWI_Master_Stop()':
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:349:3: error: 'PORT_USI' was not declared in this scope
   PORT_USI &= ~(1 << PIN_USI_SDA); // Pull SDA low.
   ^
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:349:22: error: 'PIN_USI_SDA' was not declared in this scope
   PORT_USI &= ~(1 << PIN_USI_SDA); // Pull SDA low.
                      ^
Compiling .pio/build/uno/lib631/SimpleDHT_ID849/SimpleDHT.cpp.o
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:350:21: error: 'PIN_USI_SCL' was not declared in this scope
   PORT_USI |= (1 << PIN_USI_SCL);  // Release SCL.
                     ^
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:351:12: error: 'PIN_USI' was not declared in this scope
   while (!(PIN_USI & (1 << PIN_USI_SCL)))
            ^
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:358:9: error: 'USISR' was not declared in this scope
   if (!(USISR & (1 << USIPF))) {
         ^
/home/jeroen/.platformio/lib/TinyWireM_ID797/USI_TWI_Master.cpp:358:23: error: 'USIPF' was not declared in this scope
   if (!(USISR & (1 << USIPF))) {
                       ^
*** [.pio/build/uno/lib9ac/TinyWireM_ID797/USI_TWI_Master.cpp.o] Error 1
============================================= [FAILED] Took 1.02 seconds =============================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

edgar-bonet, 2020-07-09:
Why are you using TinyWireM? This is a Wire replacement for ATtiny microcontrollers that do not have a real I2C port. It instead relies on the “universal serial interface” (USI) to provide this functionality.

The Uno does _not_ have an USI, but it does have a regular I2C port: you should use the regular Wire library with it. Note that RTClib.cpp uses the TinyWireM library _only_ when being compiled for the ATtiny85, which is clearly not your case.

jeroen24, 2020-07-09:
hey ,
i am not using the tinywirelibary i just addit the libary.
this is the exapel code i am yousing:
RTC_DS1307 rtc;

char daysOfTheWeek[7][12] = {"Sunday", "Monday", "Tuesday", "Wednesday", "Thursday", "Friday", "Saturday"};

void setup () {
  Serial.begin(9600);

#ifndef ESP8266
  while (!Serial); // wait for serial port to connect. Needed for native USB
#endif

  if (! rtc.begin()) {
    Serial.println("Couldn't find RTC");
    Serial.flush();
    abort();
  }

  if (! rtc.isrunning()) {
    Serial.println("RTC is NOT running, let's set the time!");
    // When time needs to be set on a new device, or after a power loss, the
    // following line sets the RTC to the date & time this sketch was compiled
    rtc.adjust(DateTime(F(__DATE__), F(__TIME__)));
    // This line sets the RTC with an explicit date & time, for example to set
    // January 21, 2014 at 3am you would call:
    // rtc.adjust(DateTime(2014, 1, 21, 3, 0, 0));
  }

  // When time needs to be re-set on a previously configured device, the
  // following line sets the RTC to the date & time this sketch was compiled
  // rtc.adjust(DateTime(F(__DATE__), F(__TIME__)));
  // This line sets the RTC with an explicit date & time, for example to set
  // January 21, 2014 at 3am you would call:
  // rtc.adjust(DateTime(2014, 1, 21, 3, 0, 0));
}

edgar-bonet, 2020-07-09:
What do you mean by “_i just add it the libary_”? You should not need to “add” TinyWireM anywhere to compile this example. If you can't prevent platform.io to try to compile it anyway, you should ask for help in a forum dedicated to platform.io.

jeroen24, 2020-07-09:
> What do you mean by “_i just add it the libary_”? You should not need to “add” TinyWireM anywhere to compile this example. If you can't prevent platform.io to try to compile it anyway, you should ask for help in a forum dedicated to platform.io.

ok thank you for the help.

closed 2020-07-09 by jeroen24

ladyada, 2020-07-09:
@edgar-bonet this is a feature in platformio where all dependancies are compiled, even when dependancies are not referenced in code for a platform, we've let @ivankravets know, for now folks have to manually remove the unused library

ivankravets, 2020-07-09:
Sorry for this issue. Please use [lib_ignore](https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-ignore) option and skip libraries from a build process.

joshlikespics, 2020-08-11:
This provided some more clarity for me on the TinyWireM issue in VS Code:
https://github.com/platformio/platformio-core/issues/3543

Here is an example of the suggested fix in the platformio.ini file:

[env:featheresp32]
platform = espressif32
board = featheresp32
framework = arduino
lib_ignore = TinyWireM