adafruit/RTClib #178: DateTime constructor bug causing invalid DateTime objects

Harrison-O, 2020-07-23:
DateTime constructor `DateTime(uint16_t year, uint8_t month, uint8_t day, uint8_t hour, uint8_t min, uint8_t sec)` cannot handle arguments above the correct range. E.g. `DateTime myDate = DateTime(2001, 13, 32, 25, 120, 69)` will create a DateTime object that returns impossible date and time numbers. This constructor should check for arguments too high and correct using the division and modulo operators. E.g. if the month is set as 14, then the year should be increased by one and the month decreased by 12.

Harrison-O, 2020-07-23:
Currently working on this issue on my own fork. Will do a pull request when done.

edgar-bonet, 2020-07-23:
This is a duplicate of issue #127: _Check for invalid date in DateTime_.

The adopted solution was to leave the constructor as is: the user may have knowledge that the arguments are valid, in which case it is not worth paying the cost of verification. If the user does _not_ have that knowledge, he can himself check after the fact. The official [documentation of the constructor][ctr] notes:

> ## Warning
> If the provided parameters are not valid (e.g. 31 February), the constructed `DateTime` will be invalid.
> ## See also
> The `isValid()` method can be used to test whether the constructed `DateTime` is valid.

In my opinion, getting a valid/invalid information is more useful than having the constructor try to “guess” what the user meant, and silently “fixing” the arguments.

[ctr]: https://adafruit.github.io/RTClib/html/class_date_time.html#aad62a6709a1e65e6860b1efaf8c347e9

Harrison-O, 2020-07-24:
Then maybe there should be a `fixDateTime()` method? It's nuisance to have to write extra code for every program that can cause invalid DateTime objects. Furthermore, since DateTime objects are immutable, this means a new DateTime object has be be created, which is inefficient.

Harrison-O, 2020-07-30:
It's very tedious to have to write extra code for every program that can cause invalid DateTime objects. Maybe we need a `fixDateTime()` method?

drak7, 2020-07-30:
I agree that it shouldn't try to guess the correct arguments - if someone tries to create a date of July 32nd, did they actually mean August 1st? Or do they have some invalid data?
Closing as a duplicate of #127

closed 2020-07-30 by drak7