adafruit/RTClib #179: Use of String class can be unreliable for Arduino

GunZi200, 2020-07-25:
Hi I just wanted to report an issue I had and how I fixed it. I'm very curious if someone can point out what I was doing wrong because editing the library feels like a dirty fix.

### My setup

#### IDE: 
 * Visual Studio Code with Platform IO plugin (latest 4.4.0a7)

#### Hardware:
  * Arduino Mega
  * Arduino Ethernet Shield 2
  * DS1307 RTC (I2C interface)
  * Modem for internet connection

In short I am publishing data using MQTT through the Ethernet shield and using DS1307 to timestamp my data with format `DateTime::TIMESTAMP_FULL`.

Most of the time the program would abruptly freeze and at first I thought it must be an overflow somwhere (I want to note that the program/ATmega does not reset which can sometime happen when overflow occurs). I isolated it to the following line:

```c
(rtc.now().timestamp(DateTime::TIMESTAMP_FULL)).toCharArray(time_buf, 20); // <-- program hangs here
```

```c
void mqtt_publish(char* topic, float data) {
    // 1. Convert float to char*
    // 2. Get timestamp and convert it to data type char*
    // 3. Create payload
    // 4. Publish data
    // 5. Check for error is returned

    // Step 1 
    char data_buf[5];
    dtostrf(data, 5, 2, data_buf);

    char time_buf[20]; // buffer for time stamp

    // Step 2
    (rtc.now().timestamp(DateTime::TIMESTAMP_FULL)).toCharArray(time_buf, 20); // <-- program hangs here

    // Step 3
    static char *payload = (char * ) malloc(25 * sizeof(char)); // allocate memory for MQTT message.
    uint8_t message_length = (size_t) payload;                  // length of MQTT message.
    sprintf(payload, "%s ppm %s", data_buf, time_buf);          // create message.

    // Step 4 (Retain = false, QOS  = 1)
    uint8_t error = client.publish(topic, payload, message_length, false, 1);

    free(payload); // free memory

    // Step 5 (not shown since its not relevant)
}
```

All in all this is my currently working function after modifying the library:

```c
void mqtt_publish(char* topic, float data) {
    // 1. Convert float to char*
    // 2. Get timestamp and convert it to data type char*
    // 3. Create payload
    // 4. Publish data
    // 5. Check for error is returned

    // Step 1 
    char data_buf[5];
    dtostrf(data, 5, 2, data_buf); 

    // Step 2
    char time_buf[20]; // buffer for time stamp
    rtc.now().timestamp(DateTime::TIMESTAMP_FULL, time_buf);

    // Step 3
    static char *payload = (char * ) malloc(25 * sizeof(char)); // allocate memory for MQTT message.
    uint8_t message_length = (size_t) payload;                  // length of MQTT message.
    sprintf(payload, "%s ppm %s", data_buf, time_buf);          // create message.

    // Step 4 (Retain = false, QOS  = 1)
    uint8_t error = client.publish(topic, payload, message_length, false, 1);

    free(payload); // free memory
}
```

For clarity this is how the modified timestamp function looks like:

```c
void DateTime::timestamp(timestampOpt opt, char * buffer) {
  //char buffer[20];

  // Generate timestamp according to opt
  switch (opt) {
  case TIMESTAMP_TIME:
    // Only time
    sprintf(buffer, "%02d:%02d:%02d", hh, mm, ss);
    break;
  case TIMESTAMP_DATE:
    // Only date
    sprintf(buffer, "%d-%02d-%02d", 2000 + yOff, m, d);
    break;
  default:
    // Full
    sprintf(buffer, "%d-%02d-%02dT%02d:%02d:%02d", 2000 + yOff, m, d, hh, mm,
            ss);
  }
  //return String(buffer);
}
```

I tried my best to make the use of `toCharArray()` work. Sometimes when I added Serial prints around the line it hangs it would work but that was not acceptable. I found this function here https://www.arduino.cc/reference/en/language/variables/data-types/string/functions/tochararray/

I also tried to use `c_str()` to convert the `String ` timestamp to `const char *` but the MQTT library I am using requires `char *` so that wasn't going to work. https://github.com/256dpi/arduino-mqtt

GunZi200, 2020-07-25:
One question I have is why is the `String` class used instead of `char *`? Any particular reason?

edgar-bonet, 2020-07-25:
@GunZi200 wrote:
> ```cpp
> char data_buf[5];
> dtostrf(data, 5, 2, data_buf);
> ```

The second parameter to `dtostrf()` is the minimum field width. If you want it to be 5, then the buffer should be at least 6&nbsp;bytes long (or more, depending on the value of the number). Here you have a guaranteed buffer overflow, after which the behavior of the program becomes unpredictable.

> ```cpp
> (rtc.now().timestamp(DateTime::TIMESTAMP_FULL)).toCharArray(time_buf, 20);
> ```

The method `toCharArray()` creates an extra copy of the string. You can avoid it by using `c_str()` instead to directly use the String's internal buffer. But beware that the String object then must remain valid for as long as you need that internal buffer, which means you cannot use a temporary like here.

> ```cpp
> static char *payload = (char * ) malloc(25 * sizeof(char));
> ```

You can easily avoid dynamic allocation here: `char payload[25];`.

> ```cpp
> uint8_t message_length = (size_t) payload;
> ```

This doesn't do what you thing. It converts the _address_ of the buffer (i.e. the value of the pointer) into a number. This will be much larger than the actual available space, and you risk another buffer overflow.

> this is how the modified timestamp function looks like [...]

If you want a timestamp as a `char` array, I suggest you use the `toString()` method instead.

> One question I have is why is the `String` class used instead of `char * `? Any particular reason?

This method was contributed by @AlexTB (pull request #40). It was his choice.

GunZi200, 2020-07-25:
Wow thanks for this feedback! I will try all the things you've mentioned tomorrow and report back.

GunZi200, 2020-07-25:
Well I couldn't wait to test this more and can confirm that the code below works without modifying the library :D

```c
void mqtt_publish(char* topic, float data) {
    // 1. Convert float to char*
    // 2. Get timestamp and convert it to data type char*
    // 3. Create payload
    // 4. Publish data
    // 5. Check for error if returned

    // Step 1: Convert float to char*
    char data_buf[7];
    dtostrf(data, 6, 2, data_buf);

    // Step 2: Get timestamp and convert it to data type char*.
    String time_stamp2 = rtc.now().timestamp(DateTime::TIMESTAMP_FULL);
    const char* time_stamp = time_stamp2.c_str();

    // Step 3. Create payload
    char payload[32];
    sprintf(payload, "%s ppm %s", data_buf, time_stamp);

    // 4. Publish data (Retain = false, QOS = 1)
    uint8_t error = client.publish(topic, (const char *) payload, false, 1);

    // 5. Check for error if returned.
    if (error == 0) {
        Serial.flush();
        read_return_error(client.lastError()); // Read error type.
        digitalWrite(4, LOW);
        write_to_sd_card(payload);             // Write data to SD card.
        digitalWrite(4, HIGH);
    }
}
```

closed 2020-07-30 by drak7