adafruit/RTClib #191: Fix compiler warnings

maxbla, 2020-08-04:
if you check the box "show verbose output during compilation" in the Arduino IDE, verifying any library that includes RTClib will print warnings.

```
RTClib.cpp:543:7: warning: 'isPM' may be used uninitialized in this function [-Wmaybe-uninitialized]
       if (isPM) {
       ^
RTClib.cpp:468:28: note: 'isPM' was declared here
   uint8_t hourReformatted, isPM;
                            ^
RTClib.cpp:492:47: warning: 'hourReformatted' may be used uninitialized in this function [-Wmaybe-uninitialized]
         buffer[i + 1] = '0' + hourReformatted % 10;
                                               ^
RTClib.cpp:468:11: note: 'hourReformatted' was declared here
   uint8_t hourReformatted, isPM;
```

It would be better if compiling didn't print warnings, but as far as I can tell, the uninitalized values aren't being read. The warnings are just being overly careful.

Easy fix: change 
```c++
  uint8_t hourReformatted, isPM;
  if (apTag) {     // 12 Hour Mode
    if (hh == 0) { // midnight
      isPM = false;
      hourReformatted = 12;
    } else if (hh == 12) { // noon
      isPM = true;
      hourReformatted = 12;
    } else if (hh < 12) { // morning
      isPM = false;
      hourReformatted = hh;
    } else { // 1 o'clock or after
      isPM = true;
      hourReformatted = hh - 12;
    }
  }
```
to
```c++
  uint8_t hourReformatted = hh;
  bool isPM = hh >= 12;
  if (apTag) {     // 12 Hour Mode
    if (hh == 0) { // midnight
      hourReformatted = 12;
    } else if (hh <= 12) { // morning/noon
      hourReformatted = hh;
    } else { // 1 o'clock or after
      hourReformatted = hh - 12;
    }
  }
```
I can submit a PR making this change, but I'd prefer to wait until my first PR, #190 goes through.

netdudeuk, 2020-09-20:
I get a variation of these warnings when using PlatformIO for the NODEMCUV2 -

C:\Users\me\.platformio\lib\RTClib\RTClib.cpp: In member function 'char* DateTime::toString(cha
C:\Users\me\.platformio\lib\RTClib\RTClib.cpp:543:7: warning: 'isPM' may be used uninitialized 
       if (isPM) {
       ^
C:\Users\me\.platformio\lib\RTClib\RTClib.cpp:492:51: warning: 'hourReformatted' may be used un
         buffer[i + 1] = '0' + hourReformatted % 10;
                                                   ^_

The truncation was done by something in the IDE, not by myself.

Does the code still get compiled ok ?

Would maxbla's fix work in this case ?

edgar-bonet, 2020-09-20:
Compilers differ in how they analyze the code. One compiler may see a “possible” problem where another one realizes there is no actual problem, and a third one doesn't even notice there is something to worry about.

The code should compile file for either board. maxbla's fix should silence the warnings, with no measurable difference in performance.

antwal, 2020-10-27:
Hi, 

same warnings problem on PlatformIO

```
.pio/libdeps/nanoatmega328/RTClib/RTClib.cpp: In member function 'bool DateTime::operator<(const DateTime&) const':
.pio/libdeps/nanoatmega328/RTClib/RTClib.cpp:672:23: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   return (yOff + 2000 < right.year() ||
                       ^
.pio/libdeps/nanoatmega328/RTClib/RTClib.cpp:673:24: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
           (yOff + 2000 == right.year() &&
                        ^
.pio/libdeps/nanoatmega328/RTClib/RTClib.cpp: In member function 'bool DateTime::operator==(const DateTime&) const':
.pio/libdeps/nanoatmega328/RTClib/RTClib.cpp:696:24: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   return (right.year() == yOff + 2000 && right.month() == m &&
                        ^
```

edgar-bonet, 2020-10-30:
@antwal: The `-Wsign-compare` warnings have been fixed by pull request #196.

caternuson, 2021-11-02:
Should be fixed with latest library release.

closed 2021-11-02 by caternuson