adafruit/RTClib #192: Example toString 'MM' and 'MMM' does not work correctly

pro-bowl, 2020-08-13:
Using format 'DD.MM.YYYY' in your format string MM shows double number of month (example: August prints 88)
Using format 'DD.MMM.YYYY' in your format string MMM cuts serial output after day (example: 13.08.2020 prints 13.)

drak7, 2020-08-13:
Can you provide the code you're using? I'm not seeing the same result.

pro-bowl, 2020-08-13:
I forgot to mention that I use an ESP32 Heltec Wifi Kit32
Code 'toString' from examples is unchanged except the baud rate

drak7, 2020-08-13:
I'm not seeing the same thing on an ESP32, it's working fine. We'll need as much info as you can provide to find out what's happening.
What environment are you using (Arduino, platform.io, etc)?
Which RTC are you using?
How are you setting the date/time on the RTC?

edgar-bonet, 2020-08-13:
@pro-bowl: Could you try the minimal example below?

```cpp
#include <RTClib.h>

void setup() {
    Serial.begin(9600);
    DateTime dt(2020, 8, 13);
    char buffer[] = "MM-DD-YYYY";
    Serial.println(dt.toString(buffer));
}

void loop() {}
```

Also, what version of RTClib are you using?

pro-bowl, 2020-08-13:
Sorry my fault, I found the problem. I was using another library with the DS3231 before and didn't reset the time with this lib.
Thanks for your tips.

closed 2020-08-13 by pro-bowl