adafruit/RTClib #193: alarm_mode DS3231_A1_Day or DS3231_A2_Day bug

Shuptuu, 2020-09-19:
Hi,
The "Day" mode for alarm2 or alarm1 of a DS3231 doesn't work (i.e h,m,s and day of the week matching).
And I discovered why:
in the library, days of week are considered from 0 to 6, 0 meaning Sunday.
But according the DS3231 datasheet, they must be from 1 to 7 (page 11 in https://datasheets.maximintegrated.com/en/ds/DS3231.pdf)
Also: "_The day-of-week register increments at midnight. Values that correspond to the day of week are user-defined but
must be sequential (i.e., if 1 equals Sunday, then 2 equals Monday, and so on). Illogical time and date entries result
in undefined operation._"
In other words, setting 0 for the day leads to an undefined operation: there's no automatic incrementation.
But this is exactly which is done: in the "adjust" function, line 1397 in RCTLib.cpp, the day register is always set to 0.
So of course, if you set an alarm in the "day" mode, it will never work cause the day mechanism is not set properly.

The fix (tested and validated):
from line 1391:
```
void RTC_DS3231::adjust(const DateTime &dt) {
  Wire.beginTransmission(DS3231_ADDRESS);
  Wire._I2C_WRITE((byte)DS3231_TIME); // start at location 0
  Wire._I2C_WRITE(bin2bcd(dt.second()));
  Wire._I2C_WRITE(bin2bcd(dt.minute()));
  Wire._I2C_WRITE(bin2bcd(dt.hour()));
  // Wire._I2C_WRITE(bin2bcd(0));
  Wire._I2C_WRITE(bin2bcd(dt.dayOfTheWeek()));
  Wire._I2C_WRITE(bin2bcd(dt.day()));
  Wire._I2C_WRITE(bin2bcd(dt.month()));
  Wire._I2C_WRITE(bin2bcd(dt.year() - 2000));
  Wire.endTransmission();

  uint8_t statreg = read_i2c_register(DS3231_ADDRESS, DS3231_STATUSREG);
  statreg &= ~0x80; // flip OSF bit
  write_i2c_register(DS3231_ADDRESS, DS3231_STATUSREG, statreg);
}
```
and from line 577
```
uint8_t DateTime::dayOfTheWeek() const {
  uint16_t day = date2days(yOff, m, d);
  // return (day + 6) % 7; // Jan 1, 2000 is a Saturday, i.e. returns 6
  return ((day + 6) % 7)+1; // Jan 1, 2000 is a Saturday, i.e. returns 7
}
```
Could the library be updated including this fix?

robsonos, 2020-09-20:
I came across this problem recently as well and I confirm that @Shuptuu 's solution work. All modes from `Ds3231Alarm1Mode` and `Ds3231Alarm2Mode` seem to work well after these changes. I haven't tested this changes with the PCF8523 and DS1307 modules though.

edgar-bonet, 2020-09-20:
The [documentation of the `dayOfTheWeek()`][dow] method states:

> **Returns**  
> &nbsp;&nbsp;Day of week as an integer from 0 (Sunday) to 6 (Saturday). 

The proposed fix would change this interface. This is called a “_breaking change_”, because it would break sketches that rely on this documented behavior, and work perfectly fine with the currently released version. A breaking change is a no-no for a library many people rely on.

I think the right approach is to keep the current numbering (0–6) in the library, and translate to a different numbering scheme when writing to the RTC. For example:

| day      | library | RTC |
|----------|:-------:|-----|
| Sunday   |    0    |  7  |
| Monday   |    1    |  1  |
| ...      |   ...   | ... |
| Saturday |    6    |  6  |

A simpler scheme would be to add 1 when writing to the RTC (Sunday = 1), but I would prefer Sunday to be 7, for consistency with the ISO date standard.

My untested proposal would be:

```diff
diff --git a/RTClib.cpp b/RTClib.cpp
index cee354d..709efd6 100644
--- a/RTClib.cpp
+++ b/RTClib.cpp
@@ -1389,12 +1389,13 @@ bool RTC_DS3231::lostPower(void) {
 */
 /**************************************************************************/
 void RTC_DS3231::adjust(const DateTime &dt) {
+  uint8_t dow = dt.dayOfTheWeek();
   Wire.beginTransmission(DS3231_ADDRESS);
   Wire._I2C_WRITE((byte)DS3231_TIME); // start at location 0
   Wire._I2C_WRITE(bin2bcd(dt.second()));
   Wire._I2C_WRITE(bin2bcd(dt.minute()));
   Wire._I2C_WRITE(bin2bcd(dt.hour()));
-  Wire._I2C_WRITE(bin2bcd(0));
+  Wire._I2C_WRITE(bin2bcd(dow == 0 ? 7 : dow)); // make Sunday = 7
   Wire._I2C_WRITE(bin2bcd(dt.day()));
   Wire._I2C_WRITE(bin2bcd(dt.month()));
   Wire._I2C_WRITE(bin2bcd(dt.year() - 2000));
```

[dow]: https://adafruit.github.io/RTClib/html/class_date_time.html#a958229f0d6a46d092658028d25a9e65c

commit 11ee4b2b (2020-09-22): Fix weekly alarms on the DS3231

For the `DS3231_A1_Day' and `DS3231_A2_Day' alarm modes to work, the RTC
has to know the day of the week, but `adjust()' would write 0 to the
relevant register.

Fix these alarm modes by telling the RTC the day of the week in a format
suitable to it: from 1 (Monday) to 7 (Sunday).

Fixes #193.
closed 2020-09-22 by drak7