adafruit/RTClib #197: pcf8523.ino lacks calibration example

Gambalunga, 2020-10-09:
The example sketch pcf8523.ino lacks a calibration example for the PCF8523 Real_Time Clock.

Note that in the suggested example that I give below the rather complex method of calculating the calibration offset can be reduced to the use of a constant: offset = deviation x constant (where deviation is in seconds/day (gain or loss). The resulting offset must an integer (nearest whole number)

Suggested variation for pcf8523.ino

> // Date and time functions using a PCF8523 RTC connected via I2C and Wire lib
> #include "RTClib.h"
> 
> RTC_PCF8523 rtc;
> 
> char daysOfTheWeek[7][12] = {"Sunday", "Monday", "Tuesday", "Wednesday", "Thursday", "Friday", "Saturday"};
> 
> void setup () {
>   Serial.begin(57600);
> 
> #ifndef ESP8266
>   while (!Serial); // wait for serial port to connect. Needed for native USB
> #endif
> 
>   if (! rtc.begin()) {
>     Serial.println("Couldn't find RTC");
>     Serial.flush();
>     abort();
>   }
> 
>   if (! rtc.initialized() || rtc.lostPower()) {
>     Serial.println("RTC is NOT initialized, let's set the time!");
>     // When time needs to be set on a new device, or after a power loss, the
>     // following line sets the RTC to the date & time this sketch was compiled
>     rtc.adjust(DateTime(F(__DATE__), F(__TIME__)));
>     // This line sets the RTC with an explicit date & time, for example to set
>     // January 21, 2014 at 3am you would call:
>     // rtc.adjust(DateTime(2014, 1, 21, 3, 0, 0));
>     //
>     // Note: allow 2 seconds after inserting battery or applying external power
>     // without battery before calling adjust(). This gives the PCF8523's
>     // crystal oscillator time to stabilize. If you call adjust() very quickly
>     // after the RTC is powered, lostPower() may still return true.
>   }
> 
>   // When time needs to be re-set on a previously configured device, the
>   // following line sets the RTC to the date & time this sketch was compiled
>   // rtc.adjust(DateTime(F(__DATE__), F(__TIME__)));
>   // This line sets the RTC with an explicit date & time, for example to set
>   // January 21, 2014 at 3am you would call:
>   // rtc.adjust(DateTime(2014, 1, 21, 3, 0, 0));
> 
>   // When the RTC was stopped and stays connected to the battery, it has
>   // to be restarted by clearing the STOP bit. Let's do this to ensure
>   // the RTC is running.
>   rtc.start();
> 
>   // The PCF8523 can be calibrated for:
>   //        - Aging adjustment
>   //        - Temperature compensation
>   //        - Accuracy tuning
>   // The offset mode to use, once every two hours or once every minute.
>   // The offset Offset value from -64 to +63. See the Application Note for calculation of offset values.
>   // https://www.nxp.com/docs/en/application-note/AN11247.pdf
>   // A very close approximation of the required offset for PCF8523_TwoHour (Mode 0) can be obtained with the formula offset = deviation * 2.6668
>   // The constant for offset PCF8523_OneMinute (Mode 1) is 2.8445. offset = deviation * 2.8445
>   // where deviation is in seconds per day. For accuracy the variation per day should be calculated over about 1 week.
>   // Set offset: Example - RTC gaining 6.1429 seconds per day (43 sec/week) = Offset mode PCF8523_TwoHours, and offset 16
>   // rtc.calibrate(PCF8523_TwoHours,16);  // Where mode is PCF8523_TwoHours or PCF8523_OneMinute  and offset is between - 64 and +63 
> }
> 
> void loop () {
>   DateTime now = rtc.now();
> 
>   Serial.print(now.year(), DEC);
>   Serial.print('/');
>   Serial.print(now.month(), DEC);
>   Serial.print('/');
>   Serial.print(now.day(), DEC);
>   Serial.print(" (");
>   Serial.print(daysOfTheWeek[now.dayOfTheWeek()]);
>   Serial.print(") ");
>   Serial.print(now.hour(), DEC);
>   Serial.print(':');
>   Serial.print(now.minute(), DEC);
>   Serial.print(':');
>   Serial.print(now.second(), DEC);
>   Serial.println();
> 
>   Serial.print(" since midnight 1/1/1970 = ");
>   Serial.print(now.unixtime());
>   Serial.print("s = ");
>   Serial.print(now.unixtime() / 86400L);
>   Serial.println("d");
> 
>   // calculate a date which is 7 days, 12 hours and 30 seconds into the future
>   DateTime future (now + TimeSpan(7, 12, 30, 6));
> 
>   Serial.print(" now + 7d + 12h + 30m + 6s: ");
>   Serial.print(future.year(), DEC);
>   Serial.print('/');
>   Serial.print(future.month(), DEC);
>   Serial.print('/');
>   Serial.print(future.day(), DEC);
>   Serial.print(' ');
>   Serial.print(future.hour(), DEC);
>   Serial.print(':');
>   Serial.print(future.minute(), DEC);
>   Serial.print(':');
>   Serial.print(future.second(), DEC);
>   Serial.println();
> 
>   Serial.println();
>   delay(3000);
> }

edgar-bonet, 2020-10-09:
I think [the official documentation of `RTC_PCF8523::calibrate()`][calibrate] (source [here][src]) would be a better place for at least some of this material. Although a short note in the example sketch is a good way to raise awareness.

Note that there are many possible units for a clock drift rate (s/day, s/week, min/month...), but ppm (parts per million) seems the more standard. It is used both [in the library][adjustDrift] and in the application note you linked to. Seconds per day seems more “intuitive” at first, but it is not very practical unless you measure the drift over a time span that happens to be an integer number of days.

For the sake of unit-correctness, I would write the offset formula as

> offset = round(deviation / (4.34 ppm))

[calibrate]: https://adafruit.github.io/RTClib/html/class_r_t_c___p_c_f8523.html#a33e8ae8a0e1ab3e11c4d0dc07c45d1a1
[src]: https://github.com/adafruit/RTClib/blob/1.11.3/RTClib.cpp#L1338-L1346
[adjustDrift]: https://adafruit.github.io/RTClib/html/class_r_t_c___micros.html#a9e5c1b798e0e61058c86adb74e41b6b4

Gambalunga, 2020-10-09:
Thank you for your comments Edgar.
Although I spent a lifetime in IT I am rather rusty with my programming skills and a beginner at C, therefore bound to make some syntax errors. 
I thought it may be best as a "Pull request" but I was left a bit unsure as to how it should be done.
s/day is quite easy for someone to work out who does not have access to sophisticated equipment and can be calculated exactly over an exact period of time however if the drift is checked at about the same time, give or take a few minutes after x days, it is unlikely to affect the accuracy of the calculation as the time given by the RTC is always going to be to the last (past) second.
Please feel free to alter and copy as appropriate.

Thanks

edgar-bonet, 2020-10-10:
> offset = deviation * 2.8445 where deviation is in seconds per day

Maybe it's my background as a physicist, but I am somewhat annoyed when I see a formula that misses units,¹ where the correct numerical result can only be obtained with a specific choice of units. See e.g. [Handling physical quantities in math formulas][nayuki] for a lengthy explanation.

I propose the following:

1. Update the documentation of the `calibrate()` method as follows:

------------------------------------------------------------------------

Compensate the drift of the RTC.

This method sets the "offset" register of the PCF8523, which can be used to correct a previously measured drift rate. Two correction modes are available:

- **PCF8523\_TwoHours**: Clock adjustments are performed on `offset` consecutive minutes every two hours. This is the most energy-efficient mode.

- **PCF8523\_OneMinute**: Clock adjustments are performed on `offset` consecutive seconds every minute. Extra adjustments are performed on the last second of the minute is `abs(offset)>60`.

The `offset` parameter sets the correction amount in units of roughly 4&nbsp;ppm. The exact unit depends on the selected mode:

|  mode               | offset unit                            |
|---------------------|----------------------------------------|
| `PCF8523_TwoHours`  | 4.340 ppm = 0.375 s/day = 2.625 s/week |
| `PCF8523_OneMinute` | 4.069 ppm = 0.352 s/day = 2.461 s/week |

See the accompanying sketch pcf8523.ino for an example on how to use this method.

**Parameters**

**mode** Correction mode, either `PCF8523_TwoHours` or `PCF8523_OneMinute`.  
**offset** Correction amount, from -64 to +63. A positive offset makes the clock slower.

------------------------------------------------------------------------

2. Add the following to the `setup()` in the example sketch pcf8523.ino:

```c++
// In order to improve accuracy, we can calibrate the clock drift.
// Assuming we measured a drift of 28 seconds over a period of 5 days,
// this could be corrected as follows:
const float day = 3600.0 * 24;         // one day in seconds
const float drift = 28.0 / (5 * day);  // 28 seconds in 5 days
const float drift_unit = 4.340e-6;     // 4.340 ppm
rtc.calibrate(PCF8523_TwoHours, round(drift / drift_unit));
```

What do you think? Would this properly address this issue?

@Gambalunga: I wrote “A positive offset makes the clock slower”, as this is my understanding from reading the (not so explicit) datasheet. Can you confirm that this sentence is correct?

¹ Admittedly, drift rates are dimensionless numbers. However, since they are very small (like, e.g., 6.5&nbsp;×&nbsp;10<sup>−5</sup>), it turns out to me more convenient to assign them dimensionless units such as “parts per million”, “seconds per week”, etc.

[nayuki]: https://www.nayuki.io/page/handling-physical-quantities-in-math-formulas

Gambalunga, 2020-10-11:
> 4 ppm. The exact unit depends on the selected mode:
>    1. Add the following to the setup() in the example sketch pcf8523.ino:
> are very small (like, e.g., 6.5 × 10−5), it turns out to me more

I see where you are coming from. That is to make the formula and the units
more consistent with the documentation. Unfortunately the formula, as is,
lacks something as the result is not correct. I need to re-read the
application sheet but from memory the ppm factor that they give relates to
the frequency of the crystal and that needs to be factored into the
formula. It was this complication that led me to work out a constant based
on the number of seconds per day of drift.

I will try to work out something that makes it more consistent with the
PCF8523 documentation and taking into consideration your comments.

You are correct that a positive offset makes the clock slower.

Regards

edgar-bonet, 2020-10-11:
> That is to make the formula and the units more consistent with the documentation.

It is not only that: it is also, and mostly, a matter of making the formula _correct_ with respect to the use of units. Let me give an example that, although completely unrelated, may be easier to understand: I may write the voltage across a 1&nbsp;kΩ resistor as

V = (1 kΩ) × I

where I is the current across the resistor. If I apply the formula for a current I&nbsp;= 20&nbsp;mA, I get

V = (1 kΩ) × (20 mA) = 20 kΩ⋅mA = 20 V

If I instead use I&nbsp;= 0.02&nbsp;A, I get

V = (1 kΩ) × (0.02 A) = 0.02 kΩ⋅A = 0.02 kV = 20 V

It is the same result. The formula works no matters what unit I choose for the current, as long as I take care to handle the units properly.

Now, if I forget to state the units in the original formula:

V = 1 × I

and I apply it to I&nbsp;= 20&nbsp;mA, I get

V = 20 mA

which is obviously wrong. The unit is an essential part of the formula, because the resistance is not 1 (nor 1000): it is 1&nbsp;**kΩ** or 1000&nbsp;**Ω**. The unit is part of the quantity. It is not optional.

Coming back to your formula, the correct conversion factor is not 2.6668, it is 2.6668&nbsp;**days per second**, for the same reason that a resistance cannot be 1000.

Physics school teachers spend a significant amount of time trying to get their pupils understand that units matter, and that formulas and quantities should be written with the proper units whenever relevant. Providing learning-material that goes against this principle is anti-pedagogical.

> the formula, as is, lacks something as the result is not correct.

Both your formula and mine give the same result. For 28&nbsp;seconds in 5&nbsp;days:

* your formula: 28.0 / 5 * 2.6668 = 14.934
* my formula: 28.0 / (5.0 * 86400) / 4.340e-6 = 14.934

> the ppm factor that [the application note] gives relates to the frequency of the crystal

Indeed, as the intended audience presumably has the necessary equipment for directly measuring the crystal frequency. And measuring the frequency is the easiest and fasted way of figuring out the correction factor.

> and that needs to be factored into the formula.

No, it doesn't. The correction factor only depends on the _relative_ frequency error. You can compute this error by determining the period of one oscillation (as suggested in the application note), or the time taken by twenty billion oscillations (a one week-long calibration). It does not matter, because you are interested in a _relative_ error. Measuring the drift over several days is more convenient if you do not have a frequency meter.

Gambalunga, 2020-10-13:
The original intent of calculating the constant for seconds per day was an attempt to try to keep it simple for someone who could not be bothered to work their way through, and comprehend, the PCF8523 documentation. 

In view of the above comments, and the need to conform, as far as possible, with the technical documentation perhaps the following (to be inserted in setup() section of the sketch) would suffice:
```
  // The PCF8523 can be calibrated for:
  //        - Aging adjustment
  //        - Temperature compensation
  //        - Accuracy tuning
  // The offset mode to use, once every two hours or once every minute.
  // The offset Offset value from -64 to +63. See the Application Note for calculation of offset values.
  // https://www.nxp.com/docs/en/application-note/AN11247.pdf
  // The deviation in parts per million can be calculated over a period of observation. Both the drift (which can be negative)
  // and the observation period must be in seconds. For accuracy the variation per day should be calculated over about 1 week.
  // Note: any previous calibration should cancelled prior to any new observation period.
  // Example - RTC gaining 43 seconds in 1 week
  float drift = 43; // seconds plus or minus over oservation period - set to 0 to cancel previous calibration.
  float period_sec = (7 * 86400);  // total obsevation period in seconds (86400 = seconds in 1 day:  7 days = (7 * 86400) seconds )
  float deviation_ppm = (drift / period_sec * 1000000); //  deviation in parts per million (μs)
  float drift_unit = 4.34; // use with offset mode PCF8523_TwoHours
  // float drift_unit = 4.069; //For corrections every min the drift_unit is 4.069 ppm (use with offset mode PCF8523_OneMinute)
  int offset = round(deviation_ppm / drift_unit);
  // rtc.calibrate(PCF8523_TwoHours, offset); // Un-comment to perform calibration once drift (seconds) and observation period (seconds) are correct

  Serial.print("Offset is "); Serial.println(offset); // Print to control offset
```
@edgar-bonet . I don't believe the variables need to be constants as they are only used within the scope of the calculation of the offset value. On consideration I think it is important to make it clear that the observation period can be any period (days, hours, minutes and seconds) provided it is recorded or calculated in seconds.

edgar-bonet, 2020-10-13:
I do like this much better. I still think some of this material should live in the [reference documentation][ref], but I guess it's not an issue if it is duplicated.

Minor nitpicking: although somewhat inconsistent, the formatting of the examples keeps comment lines mostly within 100&nbsp;columns. You are using 148&nbsp;columns here.

Would you submit a pull request with this? I would then submit a separate one with the update to the documentation.

> I don't believe the variables need to be constants as they are only used within the scope of the calculation of the offset value.

I assume the compiler is smart enough to do [constant propagation][fold] event without the `const` keyword. It then should make no difference on the compiled code. Using `const` is mostly a matter of clarity, to express the notion that some “variables” are not meant to ever vary. It's mostly for the benefit of the human reader.

> I think it is important to make it clear that the observation period can be any period [...] provided it is recorded or calculated in seconds.

It doesn't matter what unit you use: seconds, [centons][], whatever. As long as you use the _same_ unit for the measured offset and the measurement period, you should get the drift rate right.

[ref]: https://adafruit.github.io/RTClib/html/class_r_t_c___p_c_f8523.html#a33e8ae8a0e1ab3e11c4d0dc07c45d1a1
[fold]: https://en.wikipedia.org/wiki/Constant_folding
[centons]: https://en.battlestarwikiclone.org/wiki/Centon

Gambalunga, 2020-10-14:
Pull Request created.

closed 2020-10-15 by Gambalunga