adafruit/RTClib #204: 'hourReformatted' may be used uninitialized in this function [-Wmaybe-uninitialized]

DeeGee2010, 2020-11-02:
We resumed Standard Time yesterday, but the RTC is not updating, it's one hour ahead.

Also, my 4 digit 7 segment display is still in 24 hour mode. I believe the two following error messages go to that:

'hourReformatted' may be used uninitialized in this function [-Wmaybe-uninitialized]
'isPM' may be used uninitialized in this function [-Wmaybe-uninitialized]

My code follows:

#include <Arduino.h>
#include <Wire.h>
#include "RTClib.h"
#include <TM1637Display.h>
// Define the connections pins:
#define CLK 21
#define DIO 22
// Create rtc and display object:
RTC_DS3231 rtc;
#define SDA 14
#define SCL 13

TM1637Display display = TM1637Display(CLK, DIO);
const uint8_t blank[] = {0x00, 0x00, 0x00,0x00};

void setup() {
  // Begin serial communication at a baud rate of 9600:
  Serial.begin(9600);
  // Wait for console opening:
  delay(3000);
  Wire.begin(SDA,SCL);
  // Check if RTC is connected correctly:
  if (! rtc.begin()) {
    Serial.println("Couldn't find RTC");
    while (1);
  }
  // Check if the RTC lost power and if so, set the time:
  if (rtc.lostPower()) {
    Serial.println("RTC lost power, lets set the time!");
    // The following line sets the RTC to the date & time this sketch was compiled:
    //rtc.adjust(DateTime(F(__DATE__), F(__TIME__)));
    // This line sets the RTC with an explicit date & time, for example to set
    // January 21, 2014 at 3am you would call:
   rtc.adjust(DateTime(2020, 11, 01, 17, 27, 30));
  }
  
  display.setBrightness(0);  // Set the display brightness 
  display.setSegments(blank);//clear display
}
void loop() {
  
  DateTime now = rtc.now(); // Get current date and time
  int displaytime = (now.hour() * 100) + now.minute(); // Create time format to display
  Serial.println(displaytime); // Print displaytime to the Serial Monitor: 
    display.showNumberDecEx(displaytime,(0x80 >> 1), true); //Display the current time in 24 hour format with leading zeros enabled and a center colon:
    delay(1000);
    display.showNumberDec(displaytime, true);
    delay(1000);
    display.showNumberDec(displaytime, true); //Display the current time in 24 hour format with leading zeros enabled and a center colon:
}

edgar-bonet, 2020-11-02:
> We resumed Standard Time yesterday

I resumed Standard Time eight days ago.

> the RTC is not updating

Of course it is not. The dates for switching DST on and off are locale dependent, and sometimes change randomly due to unpredictable political decisions. For example, [on this year alone][tzdata]:

- Palestine ends DST earlier than predicted, on 2020-10-24.
- Fiji starts DST later than usual, on 2020-12-20.
- Morocco springs forward on 2020-05-31, not 2020-05-24.

Daylight Saving Time is a complex issue, and RTClib does not support it. See for example:

- issue #144: Leap year and daylight saving time
- pull request #36: Add European Daylight Saving Time / Summertime

> my 4 digit 7 segment display is still in 24 hour mode

The code you show does not attempt to display the time in 12&nbsp;hour mode, so this is normal behavior.

> 'hourReformatted' may be used uninitialized in this function [-Wmaybe-uninitialized]  
> 'isPM' may be used uninitialized in this function [-Wmaybe-uninitialized]

Duplicate of issue #191: Fix compiler warnings. The warnings are harmless, and unrelated to the “24&nbsp;hour mode”.

[tzdata]: https://launchpad.net/debian/+source/tzdata/+changelog

DeeGee2010, 2020-11-02:
Thank you. I thought the RTC (DS3231) device did more than I thought. I thought it actually connected to a NTP via radio or something.

closed 2020-11-02 by DeeGee2010