adafruit/RTClib #207: RTC_DS1307 Hang on Adafruit Grand Central M4 (NeoPixel issue?)

gcormier, 2020-11-15:
I'm having issues updating the time while in the loop.

I tried to distill it to the minimum amount of code needed, but it works OK. Adding the NeoPixel library seems to cause some issues.

Code is below, if you could confirm it is hanging for you as well?

```
#include <Adafruit_NeoPixel.h>
#include "ledscore.h"
#include <AceButton.h>
#include "RTClib.h"
using namespace ace_button;

enum class State {
  OFF,
  CLOCK_START,
  CLOCK_WAIT,
  GAME
};

void showDigit(byte position, byte value, uint32_t color);
void showAlpha(byte position, char value, uint32_t color);
void demoLoop();
void showTime();

void scoreHome(byte score, uint32_t color);
void scoreAway(byte score, uint32_t color);
void showTime(byte hh, byte mm, uint32_t color);
void updateTime();
/*
   AAA 
  F   B
  F   B
  F   B
   GGG  
  E   C
  E   C
  E   C
   DDD  
*/

#define PIN_HOME 4
#define PIN_HOME2 5
#define PIN_AWAY2 6
#define PIN_AWAY 3

#define PIN_BEEP 20

#define PIN_PIXEL 8
#define PIXEL_PER_SEGMENT 6
#define NUMPIXELS 42 * 4
#define DELAYVAL 32

#define DELAY_DEBOUNCE 10
#define DELAY_CLICK 200
#define DELAY_LONGPRESS 500
#define DELAY_DOUBLECLICK 200

#define CLOCK_CONFIG_TIME 3000 // How long to wait around at the start before leaving clock set mode

#define MAXBRIGHT 64

#define SEG_A 0
#define SEG_B PIXEL_PER_SEGMENT * 1
#define SEG_C PIXEL_PER_SEGMENT * 2
#define SEG_D PIXEL_PER_SEGMENT * 3
#define SEG_E PIXEL_PER_SEGMENT * 4
#define SEG_F PIXEL_PER_SEGMENT * 5
#define SEG_G PIXEL_PER_SEGMENT * 6

[... 571 lines cut ...]

void updateTime()
{
  now = rtc.now();
  timeHour = now.twelveHour();
  timeMin = now.minute();
}

void showTime()
{
  scoreHome(timeHour, pixels.Color(brightness, brightness, brightness));
  scoreAway(timeMin, pixels.Color(brightness, brightness, brightness));
}
```

```
[env:uno]
board = adafruit_grandcentral_m4
platform = atmelsam
framework = arduino
lib_deps = 
	adafruit/Adafruit NeoPixel
	mathertel/OneButton@0.0.0-alpha+sha.eb583d713a
	bxparks/AceButton@^1.7.0
	adafruit/RTClib@^1.12.1

ladyada, 2020-11-15:
we arent going to be able to replicate your code - please provide a minimal example. otherwise you may need to do more debugging! please re-open when you have a minimal example with *just* this library! thanks :)

closed 2020-11-15 by ladyada

gcormier, 2020-11-16:
Indeed, I think this is caused by Wire missing interrupts and hanging - NeoPixel is probably taking a lot of cycles.