adafruit/RTClib #209: RTClib does not correctly read SquareWave Pin Mode

bdlabitt, 2020-11-17:
Thank you for opening an issue on an Adafruit Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

  - **Ensure you are using an official Arduino or Adafruit board.** We can't
    guarantee a clone board will have the same functionality and work as expected
    with this code and don't support them.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  **INSERT ARDUINO BOARD NAME/TYPE HERE**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **INSERT ARDUINO
  VERSION HERE**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

`Ds3231SqwPinMode RTC_DS3231::readSqwPinMode() {
  int mode;

  Wire.beginTransmission(DS3231_ADDRESS);
  Wire._I2C_WRITE(DS3231_CONTROL);
  Wire.endTransmission();

  Wire.requestFrom((uint8_t)DS3231_ADDRESS, (uint8_t)1);
  mode = Wire._I2C_READ();

  mode &= 0x93;
  return static_cast<Ds3231SqwPinMode>(mode);`

0x93 is the wrong mask.  RS2 and RS1 are in bits 4 and 3 of the control register.  bit 2 is INTCN which can be used to see if the SQW is enabled.  Bits 1 and 0 are not related to SQW control.
Mask should be 0x9C.  Therefore the line 
`mode &= 0x93;` should be changed to
`mode &= 0x9C;`

edgar-bonet, 2020-11-17:
Rather than `0x9C` (nEOSC, RS2, RS1, INTCN), shouldn't it be `0x1C` (RS2, RS1, INTCN)? Otherwise, if the oscillator is stopped (nEOSC set), `readSqwPinMode()` would return a value that does not belong to the `enum` and is thus invalid. Also, `writeSqwPinMode()` preserves nEOSC, and one can reasonably expect that, after writing the SQW pin mode, reading it returns the value that was just written.

In order to ensure the returned value is always valid, we should also ensure that, whenever `INTCN` is set (square wave disabled), reading returns `DS3231_OFF`.

BTW, the value assigned to `DS3231_OFF` (`0x01`) seems odd. Maybe the power-on reset value of the control register (`0x1C`) would make more sense. Then this mode would not need special handling when writing.

Would you mind trying the patch below?

```diff
diff --git a/RTClib.cpp b/RTClib.cpp
index 2ecffb4..e707eae 100644
--- a/RTClib.cpp
+++ b/RTClib.cpp
@@ -1620,7 +1620,9 @@ Ds3231SqwPinMode RTC_DS3231::readSqwPinMode() {
   Wire.requestFrom((uint8_t)DS3231_ADDRESS, (uint8_t)1);
   mode = Wire._I2C_READ();
 
-  mode &= 0x93;
+  mode &= 0x1C;
+  if (mode & 0x04)
+    mode = DS3231_OFF;
   return static_cast<Ds3231SqwPinMode>(mode);
 }
 
@@ -1637,11 +1639,7 @@ void RTC_DS3231::writeSqwPinMode(Ds3231SqwPinMode mode) {
   ctrl &= ~0x04; // turn off INTCON
   ctrl &= ~0x18; // set freq bits to 0
 
-  if (mode == DS3231_OFF) {
-    ctrl |= 0x04; // turn on INTCN
-  } else {
-    ctrl |= mode;
-  }
+  ctrl |= mode;
   write_i2c_register(DS3231_ADDRESS, DS3231_CONTROL, ctrl);
 
   // Serial.println( read_i2c_register(DS3231_ADDRESS, DS3231_CONTROL), HEX);
diff --git a/RTClib.h b/RTClib.h
index 20a2464..e48960e 100644
--- a/RTClib.h
+++ b/RTClib.h
@@ -287,7 +287,7 @@ public:
 
 /** DS3231 SQW pin mode settings */
 enum Ds3231SqwPinMode {
-  DS3231_OFF = 0x01,            /**< Off */
+  DS3231_OFF = 0x1C,            /**< Off */
   DS3231_SquareWave1Hz = 0x00,  /**<  1Hz square wave */
   DS3231_SquareWave1kHz = 0x08, /**<  1kHz square wave */
   DS3231_SquareWave4kHz = 0x10, /**<  4kHz square wave */
```

bdlabitt, 2020-11-17:
This fix works.  Just ran through the combinations and all SQW are generated and the 32K osc turns on and off as well.  Thanks!

edgar-bonet, 2020-11-17:
Thanks for reporting on your tests. I opened a pull request to address this issue: #210 (Fix reading and writing the DS3231 SQW pin mode).

commit cde1f6df (2020-11-17): Fix reading and writing the DS3231 SQW pin mode

The value returned by RTC_DS3231::readSqwPinMode() is now guaranteed to
be valid, i.e. to belong to the enum Ds3231SqwPinMode.

The value of DS3231_OFF has been changed in order to avoid handling it
as a special case when writing.

Fixes #209.
closed 2020-11-17 by drak7