adafruit/RTClib #229: now.day() returns zero (0)

Souravgoswami, 2021-05-01:
So I was trying to build one clock with my RTC and a 16x2 LCD.

When I call now.day() from datetime, it returns 0. So today, 1st May 2021, is 0th May, 2021.
Last I checked it was April 27th, and it was returning April 27th.

- Arduino board: Tried both Arduino Uno and Nano

- Arduino IDE version (found in Arduino -> About Arduino menu):  **INSERT ARDUINO
  1.8.13

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too):
  
```
#pragma GCC optimize("Os")

#define ROW 2
#define COL 16
#define SPLITTER_BLINK_DELAY 500

#include <Wire.h>
#include <RTClib.h>
#include <LiquidCrystal.h>

RTC_DS1307 rtc;
LiquidCrystal lcd(12, 11, 5, 4, 3, 2) ;

char daysOfTheWeek[7][4] = {
	"Mon", "Tue", "Wed", "Thu", "Fri", "Sat", "Sun"
} ;

char Months[12][4] = {
	"Jan", "Feb", "Mar", "Apr", "May", "Jun", "Jul", "Aug", "Sep", "Oct", "Nov", "Dec"
} ;

unsigned char centre(char *str, unsigned char len) {
	unsigned char str_len = strlen(str) ;
	if(str_len >= len) return str_len ;
	return (len - str_len) / 2 ;
}

void setup() {
	Serial.begin(57600) ;
	lcd.begin(COL, ROW) ;

	while(!rtc.begin()) {
		char *message = (char *)"No RTC Found!" ;
		lcd.setCursor(centre(message, COL), 0) ;
		lcd.print(message) ;
	}

	while(!rtc.isrunning()) {
		char *message = (char *)"RTC Not Running" ;
		lcd.setCursor(centre(message, COL), 0) ;
		lcd.print(message) ;
		rtc.adjust(DateTime(F(__DATE__), F(__TIME__))) ;
	}
}

char str[100] ;
char *splitter ;
unsigned char secondNow, lastSecond, splitterUnset ;
unsigned long blinkSplitterDelay = millis() + SPLITTER_BLINK_DELAY ;
unsigned long currT ;

void loop() {
	DateTime now = rtc.now() ;
	currT = millis() ;

	secondNow = now.second() ;

	if (lastSecond != secondNow) {
		lastSecond = secondNow ;
		splitter = (char *)":" ;
		splitterUnset = 1 ;
		blinkSplitterDelay = currT + SPLITTER_BLINK_DELAY ;
	}

	if (splitterUnset && currT > blinkSplitterDelay) {
		splitter = (char *)" " ;
		splitterUnset = 0 ;
	}

	sprintf(str, "%3s, %3s %02d", daysOfTheWeek[now.dayOfTheWeek() - 1], Months[now.month() - 1], now.day() + 1) ;
	lcd.setCursor(centre(str, COL), 0) ;
	lcd.print(str) ;

	sprintf(str, "%02d%s%02d%s%02d", now.hour(), splitter, now.minute(), splitter, secondNow) ;
	lcd.setCursor(centre(str, COL), 1) ;
	lcd.print(str) ;
}
```

Now I am using LCD, but I can omit all the code, just Serial.print `now.day()`, and it's 0.

Pius171, 2021-05-01:
Try using the string example

Souravgoswami, 2021-05-01:
The string example is totally wrong:

[image]

---

Well, so there was a problem with the RTC itself. I opened the battery, inserted that again, uploaded DS1307RTC > setTime again, and it's fine now!

closed 2021-05-01 by Souravgoswami