adafruit/RTClib #230: Support printing DateTime with Milliseconds

prestoncarman, 2021-05-09:
In our project, we find the need to print a date with milliseconds when doing logging and other activities. It would be nice if the DateTime class supported the date storage and output formatting to include milliseconds. In other languages or projects, I have seen them use a format show below to print out a date time with milliseconds.  

`YYYYMMDD hh:mm:ss[.mmm]`

Resulting date time string: 2021-05-04T11:22:30.487

edgar-bonet, 2021-05-09:
The RTClib library  is focused at supporting RTC devices and most, if not all of them, have a resolution limited to the full second. Trying to get extra resolution out of an RTC is not impossible, but it would add a very significant amount of complexity.

Then, adding a milliseconds field to `DateTime` would be a waste of RAM for most users, as no existing RTC class is capable of meaningfully filling the field.

prestoncarman, 2021-05-10:
Thanks for your quick response.

After further investigation, it turns out that what was reported to me is actually using both RTC and `millis()` to create a timestamp with milliseconds. It seems we need a completely different approach.

closed 2021-05-10 by prestoncarman