adafruit/RTClib #237: Custom TwoWire and begin()

thijstriemstra, 2021-07-31:
#231 (1.14.0) added the option to use a custom TwoWire instance, which is great.

1.13.0 RTCLib's `begin` looked like this:

```cpp
boolean RTC_DS3231::begin() {
  Wire.begin();
  Wire.beginTransmission(DS3231_ADDRESS);
  if (Wire.endTransmission() == 0)
    return true;
  return false;
}
```

it now looks like this:

```cpp
boolean RTC_DS3231::begin(TwoWire *wireInstance) {
  RTCWireBus = wireInstance;
  RTCWireBus->begin();
  RTCWireBus->beginTransmission(DS3231_ADDRESS);
  if (RTCWireBus->endTransmission() == 0)
    return true;
  return false;
}
```

In my application I use custom SDA/SCL so I choose not to use `RTC_DS3231::begin()` but use my own that uses this  `begin` call:

```cpp
  // rtc
  Wire.begin(_sdaPin, _sclPin);
  Wire.beginTransmission(DS3231_ADDRESS);
  if (Wire.endTransmission() != 0) {
    Log.warning(F("Initializing DS3231... Error!" CR));
  }
```

In the latest version (1.14.0) I can't do this because `begin` now sets a protected var that I can't set from my code:

```cpp
boolean RTC_DS3231::begin(TwoWire *wireInstance) {
  RTCWireBus = wireInstance;
  // etc..
```

I can also not use `Wire.setPins(x, y)` because the code still calls `begin` on `Wire`, which I want to do elsewhere.

So I'm stuck with 1.13.0. Any ideas how to work around this pesky situation?

cc @ubidefeo

edgar-bonet, 2021-07-31:
Maybe add a parameter to the constructor for instructing it to skip the call to `begin()`. That parameter would default to `false` for backward compatibility:

In RTClib.h:

```c++
class RTC_DS1307 {
public:
  boolean begin(TwoWire *wireInstance = &Wire,
    boolean skip_begin = false);
  // ...
};
```

In RTClib.cpp:

```c++
boolean RTC_DS1307::begin(TwoWire *wireInstance, boolean skip_begin) {
  RTCWireBus = wireInstance;
  if (!skip_begin)
    RTCWireBus->begin();
  // ...
}
```

ladyada, 2021-07-31:
we would not take either PR, its non-standard. for ESP32, folks can use i2c.setpins() https://github.com/espressif/arduino-esp32/issues/3779

closed 2021-07-31 by ladyada