adafruit/RTClib #239: RTClib on ARDUINO DUE NO MORE ON SDA1 SCL1

paolovr70, 2021-08-01:
After update to relase 14 RTClib (using ARDUINO DUE board) now it RECOGNISE RTC MODULE DS3231 ON I2C AND NOT ON I2C1 AS PREVIOUSLY. Previous version searched for RTC on I2C1 in case if ARDUINO DUE. Please fix. thanks.

drak7, 2021-08-02:
You can provide a reference to a Wire object in the DS3231 begin method, it defaults to &Wire (SDA/SCL). Give it a reference to Wire1 and it should use SDA1/SCL1. E.g. `rtc.begin(&Wire1);`

ladyada, 2021-10-24:
Wire1 is the 'default' I2C port for Due. however, our standard for libraries is to use Wire as the default but allow passing a ref to begin().
Ideally if someone wants to submit an example for Due that shows how to pass in Wire1 that would be very appreciated
we can bump to 2.0.0 to signal a breaking change.

caternuson, 2021-11-02:
The library has been busified. The fix for this as mentioned above is to call begin with the Wire reference to use and override the default, ex:
```cpp
rtc.begin(&Wire1);
```

closed 2021-11-02 by caternuson

paolovr70, 2021-11-02:
Thanks for me it works fine the Proposedrtc.begin(&Wire1);Paolo.Inviato dal mio smartphone LG

rtc.begin(&Wire1);