adafruit/RTClib #240: reboot for no reason

droidprova, 2021-08-04:
Hello,

I have a problem with one of my sketches, in which I have been using your RTClib for years. It happens that if I update the sketch using LOLIN D1 MINI LITE / LOLIN D1 R2 & mini cards, with the RTClib updated to version 1.14.0 or 1.14.1, both cards do not start but enter a continuous restart loop . Obviously if I go back to the RTClib 1.13.0 version everything will work fine again.

Thank you

droidprova, 2021-08-04:
` ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00096830
~ld
`

wilhelmzeuschner, 2021-08-13:
I encountered the same issue with the latest version.
1.13.0 works fine.

Example to crash with 1.14.1:
```
#include <RTClib.h>
#include <Wire.h>
#include "Arduino.h"

RTC_DS3231 rtc;

void setup() {
  Serial.begin(115200);
  Serial.println("rtc lib crash example");

  rtc.begin();

  Serial.println("crash with next statement:");
  int hour = rtc.now().hour();
  Serial.println("this wont appear on the serial terminal");
  Serial.println(hour);
}

void loop() {
    
}
```

Environment:

- ESP32-S2 (Arduino)
- Platform IO

```
lib_deps =
     # RECOMMENDED
     # Accept new functionality in a backwards compatible manner and patches
     adafruit/RTClib @ ^1.14.1
```

```
pio update
Updating platformio/contrib-piohome           3.3.4 @ ~3.3.4                     [Up-to-date]
Updating platformio/tool-unity                1.20500.200612 @ ~1.20500.0        [Up-to-date]
Updating platformio/tool-scons                4.40100.2 @ ~4.40100.2             [Incompatible 4.40200.0]

Platform Manager
================
Platform espressif32
--------
Updating platformio/espressif32               3.3.1                              [Up-to-date]
Updating platformio/toolchain-xtensa32        2.50200.97 @ ~2.50200.0            [Up-to-date]
Updating platformio/toolchain-esp32s2ulp      1.22851.191205 @ ~1.22851.0        [Up-to-date]
Updating platformio/toolchain-riscv-esp       1.80400.0 @ ~1.80400.0             [Up-to-date]
Updating platformio/framework-arduinoespressif32 3.10006.210326 @ ~3.10006.0        [Up-to-date]
Updating platformio/tool-esptoolpy            1.30100.210531 @ ~1.30100.0        [Up-to-date]
Updating platformio/tool-mkspiffs             2.230.0 @ ~2.230.0                 [Up-to-date]
```

Crash log:
```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400268d8
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x40c
load:0x4004c000,len:0x864
load:0x40050000,len:0x2448
entry 0x4004c184
rtc lib crash example
crash with next statement:
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

Core  0 register dump:
PC      : 0x40027520  PS      : 0x00050034  A0    
  : 0x4002b47c  A1      : 0x3ffc0300
A2      : 0x00001040  A3      : 0x00019040  A4    
  : 0x000637ff  A5      : 0x3ffc02d0
A6      : 0x00000000  A7      : 0x00000000  A8    
  : 0x00000004  A9      : 0x40029764
A10     : 0x00000001  A11     : 0x00000000  A12   
  : 0x800251fc  A13     : 0x3ffc02a0
A14     : 0x00000000  A15     : 0x3ffc02dc  SAR   
  : 0x0000001f  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x800251fc  LEND  
  : 0x3ffc02a0  LCOUNT  : 0x40026408
Core  0 was running in ISR context:
EPC1    : 0x40088c17  EPC2    : 0x00000000  EPC3  
  : 0x00000000  EPC4    : 0x40027520

Backtrace:0x4002751d:0x3ffc03000x4002b479:0x3ffc7900 0x40084287:0x3ffc7940 0x40081fd6:0x3ffc7990 0x4008134a:0x3ffc79b0 0x400813d5:0x3ffc7a00 0x400814dc:0x3ffc7a20 0x400810e7:0x3ffc7a50 0x40082b1a:0x3ffc7a90

ELF file SHA256: 0000000000000000

Rebooting...
```

Here's a decoded backtrace (although I am pretty unsure about it's validity since I am using `monitor_filters = esp32_exception_decoder` and I'm using an ESP32-S2):
(In another, more complex project, I had the same issue and then the decoder showed Wire.beginTransmission() to be the issue...)
```
Backtrace:0x4002751d:0x3ffc03000x4002b479:0x3ffc7930 0x40084287:0x3ffc7970 0x40081f80:0x3ffc79c0 0x400812a9:0x3ffc79e0 0x400813e5:0x3ffc7a00 0x400814d2:0x3ffc7a20 0x400810e7:0x3ffc7a50 0x40082b1a:0x3ffc7a90 
  #0  0x4002751d:0x3ffc03000 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105
```

platformio.ini (for transparency), known to work:
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32s2]
platform = espressif32
; toolchain download links see
; refer "name": "xtensa-esp32s2-elf-gcc","version": "gcc8_4_0-esp-2021r1" section of
; https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json
; e.g. Windows: https://github.com/espressif/crosstool-NG/releases/download/esp-2021r1/xtensa-esp32s2-elf-gcc8_4_0-esp-2021r1-win32.zip
platform_packages =
        toolchain-xtensa32s2@file://D:\Downloads\xtensa-esp32s2-elf
        ;framework-arduinoespressif32
        framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#a4118ea88987c28aac3a49bcb9cc5d6c0acc6f3f
        platformio/tool-esptoolpy @ ~1.30100
framework = arduino
board = esp32dev
board_build.mcu = esp32s2
board_build.partitions = huge_app.csv
board_build.variant = esp32s2
board_build.f_cpu = 240000000L
board_build.f_flash = 80000000L
;board_build.flash_mode = dio
board_build.arduino.ldscript = esp32s2_out.ld
build_unflags =
        -DARDUINO_ESP32_DEV
        -DARDUINO_VARIANT="esp32"
build_flags =
        -DARDUINO_ESP32S2_DEV
        -DARDUINO_VARIANT="esp32s2"

monitor_speed = 115200
lib_deps =
     # RECOMMENDED
     # Accept new functionality in a backwards compatible manner and patches
     adafruit/RTClib @ ^1.14.1
     adafruit/Adafruit NeoPixel @ ^1.8.5
```

blak3r, 2021-09-11:
Can confirm similar issue on ESP2866.  Rolling back library to 1.13.0 fixed the exceptions.

Don't have the trace anymore but it was happening in HardwareSerial.write via the now function. 
For what it's worth, this occurred in my application which is albeit not a minimal sketch and was being compiled in with platform.io.

I was able to load the example/DS1307.ino from Arduino IDE and it ran perfectly.

blak3r, 2021-09-13:
Update: I had an MPU6050 (accelerometer/gyro) on my board.  MPU6050 coincidently has the same exactly I2C address as DS1307 (0x68)! Once I set MPU6050 to be on Ox69 - I upgraded the RTC lib back to 1.14.1 and I no longer have the reboot exceptions. 

Perhaps that exception was correctly warning me that I had a conflict on the bus...

droidprova, 2021-09-16:
Thanks for the feedback, but in my case the sketch restarts even without peripherals connected. However, it must be said that in my project I have 3 objects connected to the i2c bus: 1 bme280 sensor, a ds3231 and a 128x64 oled. However, all these objects have a unique address, so the cause of the restart cannot be determined by a conflict on the bus. 

Hello and thanks

caternuson, 2021-11-02:
These look like watchdog timer resets:
```
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 
```
These are usually fixed in the core. Is this issue Platform IO specific? Anyone else getting this using the Arduino IDE and latest library and BSPs?

caternuson, 2021-11-19:
Closing due to lack of response and unable to recreate issue.

Example above runs fine on Metro ESP32-S2.
RTC Lib = 2.0.1
ESP32 BSP = 2.0.1
[image]

closed 2021-11-19 by caternuson