adafruit/RTClib #262: ATTiny Compatibility Has Been Broken for a Year Now

NoLANline, 2022-06-01:
Thank you for opening an issue on an Adafruit Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

  - **Ensure you are using an official Arduino or Adafruit board.** We can't
    guarantee a clone board will have the same functionality and work as expected
    with this code and don't support them.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  Uno as ISP for ATTiny85

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.13

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 

1. Change Arduino board settings to ATTiny85 (I used 1, 8 and 16MHz)
2. Download the latest library, or any library after 13.X.X
3. Create a blank sketch with a simple #include "RTClib.h"
4. Attempt to verify & receive a whole bunch of SPI/Wire.h compilation errors

@ubidefeo made a fatal (in regards to the Tiny's at least) PR between 13.X.X and 14.0.0; I would investigate further now, however time is of the essence to me. Perhaps at a later time.

ERRORS RECIEVED:
`
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src\utility\twi.c: In function 'twi_init':
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src\utility\twi.c:88:16: error: 'SDA' undeclared (first use in this function)
   digitalWrite(SDA, 1);
                ^~~
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src\utility\twi.c:88:16: note: each undeclared identifier is reported only once for each function it appears in
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src\utility\twi.c:89:16: error: 'SCL' undeclared (first use in this function); did you mean 'SPL'?
   digitalWrite(SCL, 1);
                ^~~
                SPL
In file included from c:\program files (x86)\arduino\hardware\tools\avr\avr\include\avr\io.h:99:0,
                 from C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src\utility\twi.c:26:
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src\utility\twi.c:92:7: error: 'TWSR' undeclared (first use in this function); did you mean 'ACSR'?
   cbi(TWSR, TWPS0);
       ^
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src\utility\twi.c:92:3: note: in expansion of macro 'cbi'
[... 213 lines cut ...]
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src\utility\twi.c:505:10: error: 'TWS3' undeclared (first use in this function); did you mean 'TWS4'?
   switch(TW_STATUS){
          ^
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src\utility\twi.c:510:7: error: 'TWDR' undeclared (first use in this function); did you mean 'TWSR'?
       TWDR = twi_slarw;
       ^~~~
       TWSR
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src\utility\twi.c:530:10: error: 'TWCR' undeclared (first use in this function); did you mean 'TWDR'?
          TWCR = _BV(TWINT) | _BV(TWSTA)| _BV(TWEN) ;
          ^~~~
          TWDR
In file included from c:\program files (x86)\arduino\hardware\tools\avr\avr\include\avr\io.h:99:0,
                 from C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src\utility\twi.c:26:
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src\utility\twi.c:530:21: error: 'TWINT' undeclared (first use in this function)
          TWCR = _BV(TWINT) | _BV(TWSTA)| _BV(TWEN) ;
                     ^
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src\utility\twi.c:530:34: error: 'TWSTA' undeclared (first use in this function); did you mean 'TWS3'?
          TWCR = _BV(TWINT) | _BV(TWSTA)| _BV(TWEN) ;
                                  ^
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src\utility\twi.c:530:46: error: 'TWEN' undeclared (first use in this function); did you mean 'TWINT'?
          TWCR = _BV(TWINT) | _BV(TWSTA)| _BV(TWEN) ;
                                              ^
exit status 1
Error compiling for board ATtiny25/45/85.
`

edgar-bonet, 2022-06-01:
What core are you using?

Could you try to compile the sketch below?

```c++
#include <Wire.h>
```

drak7, 2022-06-01:
I don't see any issues using ATTinyCore 1.5.2 targeting ATtiny25/45/85, building in Arduino IDE 1.8.19.
attiny (Mellis) does not have a Wire implementation so the compiler is trying to use the AVR one.

Please review the [code of conduct](https://github.com/adafruit/RTClib/blob/master/code-of-conduct.md) and refrain from blaming a particular contributor for your issue. It would be more constructive to mention the specific PR number that you think introduced it.

caternuson, 2022-08-04:
Closing. Also not seeing any issues compiling for an Adafruit Trinket 5V.

Test sketch:
```cpp
#include "RTClib.h"

void setup() {
}

void loop() {
}
```

closed 2022-08-04 by caternuson