adafruit/RTClib #268: DS1307 - Unable to read more than 32 bytes at a time from RAM (remaining bytes are incorrectly returned as 0xFF)

JOULTICOA, 2022-12-18:
There seems to be some limitation when reading more than 32 bytes at a time from the DS1307 RAM

Bytes 33-56 (1 index) will be returned as 0xFF no matter what

However if reading the first 32, then the last 24, it returns the correct bytes.

This is using Arduino Pro Mini

Is this a known issue / limitation?

I reviewed the code of the library and don't see anything that immediately stands out

Here is the first INO I was using, before I did a second test reading 32, then the remaining 24 separately
[ds1307 Reader.zip](https://github.com/adafruit/RTClib/files/10253357/ds1307.Reader.zip)

edgar-bonet, 2022-12-18:
Could this be a limitation of the Adafruit\_BusIO library? Adafruit\_I2CDevice.cpp has [this code][maxBufferSize]:

```c++
#ifdef ARDUINO_ARCH_SAMD
  _maxBufferSize = 250; // as defined in Wire.h's RingBuffer
#elif defined(ESP32)
  _maxBufferSize = I2C_BUFFER_LENGTH;
#else
  _maxBufferSize = 32;
#endif
```

I do not quite understand what is going on though. [The method `Adafruit_I2CDevice::read()`][read] carries a comment stating that it cannot read more than `maxBufferSize()` bytes, yet it implements a loop that reads as many bytes as requested in `maxBufferSize()`-sized chunks. It may be worth recompiling the library with `DEBUG_SERIAL` defined.

[maxBufferSize]: https://github.com/adafruit/Adafruit_BusIO/blob/1.14.1/Adafruit_I2CDevice.cpp#L14-L20
[read]: https://github.com/adafruit/Adafruit_BusIO/blob/1.14.1/Adafruit_I2CDevice.cpp#L168-L187

JOULTICOA, 2022-12-18:
> Could this be a limitation of the Adafruit_BusIO library? Adafruit_I2CDevice.cpp has [this code](https://github.com/adafruit/Adafruit_BusIO/blob/1.14.1/Adafruit_I2CDevice.cpp#L14-L20):
> 
> ```c++
> #ifdef ARDUINO_ARCH_SAMD
>   _maxBufferSize = 250; // as defined in Wire.h's RingBuffer
> #elif defined(ESP32)
>   _maxBufferSize = I2C_BUFFER_LENGTH;
> #else
>   _maxBufferSize = 32;
> #endif
> ```
> 
> I do not quite understand what is going on though. [The method `Adafruit_I2CDevice::read()`](https://github.com/adafruit/Adafruit_BusIO/blob/1.14.1/Adafruit_I2CDevice.cpp#L168-L187) carries a comment stating that it cannot read more than `maxBufferSize()` bytes, yet it implements a loop that reads as many bytes as requested in `maxBufferSize()`-sized chunks. It may be worth recompiling the library with `DEBUG_SERIAL` defined.

I think this must be it!  I'll play with these libraries (twi.h, wire.h, and the CPP you mentioned above) and see if the 328P can handle a 56 or 64 byte buffer.  If not, I guess a DUE or something would

Thanks!

closed 2022-12-18 by JOULTICOA