adafruit/RTClib #271: Problem with RTClib upgrade

Rollmops67, 2023-01-15:
Hello from France !
That's my first post here, please be indulgent !

I have an arduino program running on an Arduino Nano (Atmega328p) 24 hours a day since more than one year without problems.
On the I2C bus there are a LCD display at address 0x27 (4 lines of 20 characters) and a DS3231 clock.

As said, the program runs fine since more than one year.
Today I wanted tu update my RTClib library (version 1.13.0) to the latest version available (2.1.1)
After that, when recompiling my sketch the program doesn't work anymore and crashes about every second at the end of the setup part.

Here are the last lines in the setup part of my sketch :
-------------------------------------------
lcd.init();                               // LCD initialization
lcd.backlight();                     // LCD backlight ON
lcd.createChar(0, Finv);        // creating a special character

initecran1();                            // Subroutine to write some constantly displayed text on the screen

DateTime now = rtc.now();    // gathering of actual date and time
actual = now.day();                // memorizing the actual day

SETUP ENDS HERE
----------------------------------------------------
In fact when the program starts the LCD backlight goes on, the text of the subroutine "initecran1()" shows up on the LCD screen, and just after that the program crashes and starts again and again doing the same.

For me it looks like it's the "DateTime now = rtc.now();" command that crashes the program.

If I go back to the 1.13.0 version of RTClib all is OK again. (fortunately !)

Do I miss something (changing of the syntax of the RTC commands) or what ?

TIA for every hint !

Roland

edgar-bonet, 2023-10-24:
I don't have a solution, just a few thoughts that may help figuring this out:

1. Did you call `rtc.begin()` in your `setup()`?
2. Does this happen if you replace the DS3231 with the virtual RTC `RTC_Millis`?
3. Does this happen if you remove the LCD, and the RTC is the only device on the I2C bus?
4. Could you provide a minimal sketch that shows the faulty behavior? If the sketch can work with no extra hardware (using `RTC_Millis` and sending output to `Serial`) that would be better.
5. Are you familiar with [`git bisect`][bisect]? If so, could you bisect the problem to a specific commit?

[bisect]: https://git-scm.com/docs/git-bisect/fr

Rollmops67, 2023-10-24:
Hello from Strasbourg, Edgar !

You asked :
"Did you call rtc.begin() in your setup()?"

The answer is no !
I don't know if more than 2 years ago when I wrote my sketch this was necessary with the version of RTCLib from then.
When I wrote the sketch I took a look at the examples in this repository, so either it was not necessary then or I didn't see it in the examples. We will never know !
Right now there is indeed a `" if (! rtc.begin()) {........ "` lign in the examples...

Anyway when adding `"rtc.begin());"` now IT WORKS  with version 2.1.1!!! 🥇 

YEEAAHHHH !

The question is just why did it work without "rtc.begin());" in RTCLib versions before 1.14 ???
But OK, it's no longer important now....

Alors un grand merci à toi Edgar, ça fait un moment que je m'arrachais les cheveux sur ce problème !

Roland

edgar-bonet, 2023-10-24:
I am glad your issue is solved. :smiley:

> just why did it work without `rtc.begin();` in RTCLib versions before 1.14&nbsp;???

RTClib version 1.14.0 introduced the ability to use a custom I2C interface (see pull request #231). This can be useful if you have more than one such interface. Since then, the `begin()` method is used to select the I2C interface. Its signature is the following:

```c++
bool begin(TwoWire *wireInstance = &Wire);
```

If you don't call `begin()`, then the internal pointer held by the library to identify the interface stays uninitialized. Before version&nbsp;1.14, the I2C interface was hard-coded to `Wire`.

> So I think we can close the issue...

Indeed, you can close it (why did you reopen it?).