adafruit/RTClib #274: Constructor DateTime(const char *iso8601dateTime) is not compliant with iso8601 times.

matdon, 2023-02-09:
The constructor DateTime(const char *iso8601dateTime) claims to enable the creation of DateTime objects using an ISO8601 date time definition.

However, the constructor only works for iso datetimes of format:
_"2000-01-01T00:00:00"_ that can be truncated from the right at an arbitrary position, e.g. the string _"2000-01"_ leads to a valid result.

This can be seen from the constructor definition in the code:
`DateTime::DateTime(const char *iso8601dateTime) {
  char ref[] = "2000-01-01T00:00:00";
  memcpy(ref, iso8601dateTime, min(strlen(ref), strlen(iso8601dateTime)));
  yOff = conv2d(ref + 2);
  m = conv2d(ref + 5);
  d = conv2d(ref + 8);
  hh = conv2d(ref + 11);
  mm = conv2d(ref + 14);
  ss = conv2d(ref + 17);
}` (RTClib.cpp, line 369)

Other definitions ([https://en.wikipedia.org/wiki/ISO_8601#Times](url)) like times only, e.g. _T18:00_, do not work. It is necessary to always define a full date with year, month and day before a time can be specified. Further, the colon between hours and minutes is mandatory for the constructor to work correctly. This is not the case in the ISO standard.

edgar-bonet, 2023-02-09:
I just submitted a pull request for clarifying the documentation.

Regarding the example `"T18:00"`, note that it would not make much sense in this context, because:

 * the other fields would have to default to something, as a `DateTime` object always represents a fully specified date/time 
 * the only sensible default date would be the current date
 * the library has no notion of “current date”.

matdon, 2023-02-18:
Dear edgar-bonet, thank you for your (incredibly) fast answer and your work in the pull request.

From my point of view, this is of great help to prevent misunderstanding in the future.
As you mentioned, it does not make sense to use a DateTime class object as defined in the library for anything else than a full date at the moment. Storing only time values is not meaningful.

However, this might not be clear at first glance when looking at the current documentation, as a DateTime objects defaults to the reference:

`char ref[] = "2000-01-01T00:00:00";`

and a user (as me) would come to the idea of living with the default year, month and day and storing only time information.

Your pull request helps here.

From my point of view, the issue is closed.

However one might think in the future of allowing all ISO 8601 specifiers, and point the user to the fact that default values for not specified items are used as in the current ref string. The user then can decide on how to treat the defaults on his own.
This should be mentioned in the documentation then as well.

Best regards, Matthias

closed 2023-02-18 by matdon