adafruit/RTClib #276: RTClib with DS3231module on Arduino Uno WiFi

IFoundABugAgain, 2023-02-24:
- Arduino board:  **Arduino Uno WiFi Rev2**
- Arduino IDE version (found in Arduino -> About Arduino menu):  **2.0.3**
- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

I have an Arduino Uno WiFi Rev2 board with a DS3231 RTC module connected to the Arduino via I2C. My IDE is version 2.0.3. I am using this Lib: https://github.com/adafruit/RTClib to work with the real time clock module (the DS3231).
I have in the main loop code: Datetime tiktok = rtc.now(). This gets the time via the DS3231 RTC module via I2C and puts it into the Datetime variable "tiktok". You can then use tiktok.day() to get the day, for example the 23rd. Randomly however it will output the 3rd (when it is the 23rd) for a loop or a few loops then revert back to the 23rd. When it is the 24th it shows the 4th randomly. 

[Solar_copy_20230208220924.txt](https://github.com/adafruit/RTClib/files/10828967/Solar_copy_20230208220924.txt)

caternuson, 2023-02-24:
Can you recreate the same behavior with the example sketch from the library?
https://github.com/adafruit/RTClib/blob/master/examples/ds3231/ds3231.ino

IFoundABugAgain, 2023-02-24:
> Can you recreate the same behavior with the example sketch from the library? https://github.com/adafruit/RTClib/blob/master/examples/ds3231/ds3231.ino

I have loaded it now and will run it for a few days and see if the same occurs.

IFoundABugAgain, 2023-02-26:
Ok i can confirm the issue still happens with the test/example code. I have put it [https://forums.adafruit.com/viewtopic.php?t=199131](url) also.

`
// Date and time functions using a DS3231 RTC connected via I2C and Wire lib
#include "RTClib.h"

RTC_DS3231 rtc;

void setup () {
  Serial.begin(9600);

#ifndef ESP8266
  while (!Serial); // wait for serial port to connect. Needed for native USB
#endif

  if (! rtc.begin()) {
    Serial.println("Couldn't find RTC");
    Serial.flush();
    while (1) delay(10);
  }

  if (rtc.lostPower()) {
    Serial.println("RTC lost power, let's set the time!");
    // When time needs to be set on a new device, or after a power loss, the
    // following line sets the RTC to the date & time this sketch was compiled
    rtc.adjust(DateTime(F(__DATE__), F(__TIME__)));
    // This line sets the RTC with an explicit date & time, for example to set
    // January 21, 2014 at 3am you would call:
    // rtc.adjust(DateTime(2014, 1, 21, 3, 0, 0));
  }

  // When time needs to be re-set on a previously configured device, the
  // following line sets the RTC to the date & time this sketch was compiled
  // rtc.adjust(DateTime(F(__DATE__), F(__TIME__)));
  // This line sets the RTC with an explicit date & time, for example to set
  // January 21, 2014 at 3am you would call:
  // rtc.adjust(DateTime(2014, 1, 21, 3, 0, 0));
}
int count = 0;
int day = -1;
void loop () {
    DateTime now = rtc.now();
    Serial.print("Day with DEC: ");Serial.print(now.day(), DEC);Serial.print("  Day no DEC:  ");Serial.print(now.day());Serial.print("   UnixTime: ");Serial.print(now.unixtime());Serial.print("   Count: ");Serial.println(count);
    if (now.day() != day) {   
      day = now.day();
      count = count + 1;
      Serial.print("day var: ");Serial.print(day);Serial.print(" now.day(): ");Serial.println(now.day());
    }
    delay(3000);
}
`

OUTPUT: 

`
18:03:46.005 -> Day with DEC: 6  Day no DEC:  6   UnixTime: 1044554624   Count: 13
18:03:48.998 -> Day with DEC: 26  Day no DEC:  26   UnixTime: 1677434627   Count: 13
18:03:49.094 -> day var: 26 now.day(): 26
18:03:52.055 -> Day with DEC: 6  Day no DEC:  6   UnixTime: 1044554630   Count: 14
18:03:52.119 -> day var: 6 now.day(): 6
18:03:55.077 -> Day with DEC: 6  Day no DEC:  6   UnixTime: 1044554633   Count: 15
18:03:58.071 -> Day with DEC: 6  Day no DEC:  6   UnixTime: 1044554636   Count: 15
18:04:01.063 -> Day with DEC: 6  Day no DEC:  6   UnixTime: 1044554639   Count: 15
18:04:04.086 -> Day with DEC: 26  Day no DEC:  26   UnixTime: 1677434642   Count: 15
18:04:04.149 -> day var: 26 now.day(): 26
18:04:07.112 -> Day with DEC: 26  Day no DEC:  26   UnixTime: 1677434645   Count: 16
18:04:10.104 -> Day with DEC: 26  Day no DEC:  26   UnixTime: 1677434648   Count: 16
18:04:13.095 -> Day with DEC: 26  Day no DEC:  26   UnixTime: 1677434651   Count: 16
18:04:16.119 -> Day with DEC: 26  Day no DEC:  26   UnixTime: 1677434654   Count: 16
18:04:19.113 -> Day with DEC: 26  Day no DEC:  26   UnixTime: 1677434657   Count: 16
18:04:22.136 -> Day with DEC: 26  Day no DEC:  26   UnixTime: 1677434660   Count: 16
18:04:25.128 -> Day with DEC: 26  Day no DEC:  26   UnixTime: 1677434663   Count: 16
18:04:28.120 -> Day with DEC: 26  Day no DEC:  26   UnixTime: 1677434666   Count: 16
18:04:31.141 -> Day with DEC: 26  Day no DEC:  26   UnixTime: 1677434669   Count: 16
18:04:34.131 -> Day with DEC: 26  Day no DEC:  26   UnixTime: 1677434672   Count: 16

`

IFoundABugAgain, 2023-02-27:
The bug has been confirmed by a second user on https://forums.adafruit.com/memberlist.php?mode=viewprofile&u=337713.

It feels like an 8bit shift but how or why I have no idea.

edgar-bonet, 2023-02-27:
For what it's worth, the Unix times on the second and third lines of the output represent to the following dates:

    1677434627 → 2023-02-26 18:03:47
    1044554630 → 2003-02-06 18:03:50

Both the day-of-month and year fields were corrupted, as follows:

    buffer[4] (date): 0x26 (0b00100110) → 0x06 (0b00000110)
    buffer[6] (year): 0x23 (0b00100011) → 0x03 (0b00000011)

Both errors are single-bit flips, where the corrupted bit was originally a one that was preceded and followed by at least two zeros. The pattern is 00100 → 00000.

It looks like a hardware-level communication problem to me. What module are you using? How is it connected to the Arduino? How noisy is the electrical environment?

caternuson, 2023-03-03:
Resolved. Wiring issue:
https://forums.adafruit.com/viewtopic.php?p=963052#p963052

closed 2023-03-03 by caternuson