adafruit/RTClib #281: Issue : Should react to the case of IIC read failure （应该对IIC读取失败的情况做出反应）

MR-XieXuan, 2023-03-12:
[image]
If the read fails, you should return a DateTime error, which is not good for your program.
write_then_read returns false on failure and true on success.
The library should also handle this, such as returning to nullptr on failure.
It looks like this.
[image]

Chinese (中文) : 
如果读取失败,应该会出现返回DateTime错误的情况,这样子对程序不利。
而 write_then_read 在操作失败以后会返回 false ， 成功后会返回true。
库中也应该对这个进行处理,比如 在失败时放回 nullptr 。
像这样子。

caternuson, 2023-03-13:
What RTC module is being used?

What Arduino sketch is being run?

The validity of the I2C connection is checked in `begin()` which returns false if it fails. If `begin()` is returning `true` but I2C is failing later, then it may be something else.

MR-XieXuan, 2023-03-13:
RTC module : RTC_DS1307
Sometimes this problem occurs : 
[image]

MohammedMMC, 2024-12-16:
i have the same on MAX17048G
[image]