adafruit/RTClib #293: ESP32 crashes when .now() is called before .begin()

DiyLabsElectronicsDivision, 2023-10-18:
I'm having an issue where my ESP32 crashes when I call .now() before calling .begin(). I am using a generic ESP32 dev board, which has an ESP32-WROOM-32 chip. Arduino IDE version is 2.2.1, RTClib version is 2.1.1. I installed the RTClib library through the Arduino IDE's library manager. 

This is the code which crashes my ESP32:

`#include <RTClib.h>

RTC_DS1307 rtc;

void setup() {
  // put your setup code here, to run once:

  //When rtc.begin() is commented out, the ESP32 crashes
  //rtc.begin(); 

  DateTime time = rtc.now(); //This line causes a crash when rtc.begin() is not called

}

void loop() {
  // put your main code here, to run repeatedly:

}
`

If rtc.begin() is uncommented, everything works fine. The ESP32 prints this in the serial terminal:

`ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d1337  PS      : 0x00060030  A0      : 0x800d1419  A1      : 0x3ffc5150  
A2      : 0x00000000  A3      : 0x3ffc51a9  A4      : 0x00000001  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffc2630  
A10     : 0x400825f8  A11     : 0x00000001  A12     : 0x3ffbcca0  A13     : 0x3ffbcc80  
A14     : 0x00000000  A15     : 0x3ffc21cc  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000c  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400d1334:0x3ffc5150 0x400d1416:0x3ffc5170 0x400d12a8:0x3ffc5190 0x400d1288:0x3ffc51d0 0x400d239e:0x3ffc5200

ELF file SHA256: 31cee8c7bff2a5e3`

I suspect it's caused by a pointer being 
uninitialized, and causing garbage values to be used instead.

edgar-bonet, 2023-10-18:
It is indeed a stray pointer: [`i2c_dev` is a member of `RTC_I2C`][def] (the parent class of `RTC_DS1307`), initially `NULL`:

```c++
  Adafruit_I2CDevice *i2c_dev = NULL; ///< Pointer to I2C bus interface
```

It is properly [initialized in `RTC_DS1307::begin()`][init]:

```c++
  i2c_dev = new Adafruit_I2CDevice(DS1307_ADDRESS, wireInstance);
```

and [dereferenced in `RTC_DS1307::now()`][use]:

```c++
  i2c_dev->write_then_read(buffer, 1, buffer, 7);
```

The solution to your problem is to simply call `begin()`. Whereas it would be possible for the library to check the validity of the pointer within `now()`, that would bring its own issues:

* For consistency, every single method of `RTC_DS1307` (and of the other RTC classes) should be similarly guarded. The cost of doing so may be non trivial for a library designed to work well in very constrained environments (think Adafruit Trinket).

* If `now()` is called before `begin()`, the library has no way of reporting this error: `Serial` may well be connected to an external device that would not cope well with receiving unexpected error messages.

IMHO, in this situation it makes sense to just say that, if any method accessing the RTC is called before `begin()`, [the behavior is undefined][ub].

[def]: ../blob/2.1.1/src/RTClib.h#L341
[init]: ../blob/2.1.1/src/RTC_DS1307.cpp#L17
[use]: ../blob/2.1.1/src/RTC_DS1307.cpp#L58
[ub]: https://en.cppreference.com/w/cpp/language/ub

caternuson, 2023-10-18:
`begin()` should be called first.

closed 2023-10-18 by caternuson

DiyLabsElectronicsDivision, 2023-10-19:
Yes, I didn't think of all the issues that would come with checking the validity of a pointer. There is no reason to call `now()` before `begin()`. I spent a good half-hour going through my code trying to see what was causing the crash, before finding out that a function that I was calling was using `now()` and I hadn't noticed.

mplattu, 2025-03-13:
I ran to this issue since I got very similair guru meditation (LoadProhibited) on ESP32 and `RTC_PCF8523`. I did call `begin()` before `now()` but got the 

It turned out I was calling `now()` too often. I tried calling RTC not more often than every second and got rid of the issue.

```
#define RTC_READ_FREQUENCY_MS 1000

DateTime lastRtcNowValue;
uint32_t lastRtcNowValueInvalid;

DateTime rtcNow()
{
    if (millis() > lastRtcNowValueInvalid) {
        lastRtcNowValue = rtc.now();
        lastRtcNowValueInvalid = millis() + RTC_READ_FREQUENCY_MS;
    }

    return lastRtcNowValue;
}
```

Set `lastRtcNowValueInvalid = millis() -1` before calling `rtcNow()` for the first time. Note: This does not handle the case of `millis()` rotate back to 0.

edgar-bonet, 2025-03-13:
> This does not handle the case of `millis()` rotate back to 0.

Here is how you can easily handle this case:

```c++
#define RTC_READ_PERIOD_MS 1000

DateTime lastRtcNowValue;
uint32_t lastRtcNowMillis;

DateTime rtcNow()
{
    uint32_t currentMillis = millis();
    if (currentMillis - lastRtcNowMillis >= RTC_READ_PERIOD_MS) {
        lastRtcNowValue = rtc.now();
        lastRtcNowMillis = currentMillis;
    }
    return lastRtcNowValue;
}
```

C.f. [How can I handle the millis() rollover?][ase]. Note that I renamed `RTC_READ_FREQUENCY_MS` to the more appropriate `RTC_READ_PERIOD_MS`.

[ase]: https://arduino.stackexchange.com/questions/12587/how-can-i-handle-the-millis-rollover