adafruit/RTClib #301: Add a Timespan in an "IF" condition doesn't work

Rollmops67, 2024-04-09:
Hello all !

I'm having a problem with adding a timespan when using an "IF" condition.
Software to compile : Arduion IDE 1.8.9
Hardware : Arduino nano
RTCLib version : 2.1.3

This works :
```
#include "RTClib.h"
RTC_DS3231 rtc;

void setup ()
{
  Serial.begin(57600);
  rtc.begin();
}

loop()
{
 DateTime now = rtc.now();
 // adding an hour
    DateTime future (now + TimeSpan(0,1,0,0));

    Serial.print(future.year(), DEC);
    Serial.print('/');
    Serial.print(future.month(), DEC);
    Serial.print('/');
    Serial.print(future.day(), DEC);
    Serial.print(' ');
    Serial.print(future.hour(), DEC);
    Serial.print(':');
    Serial.print(future.minute(), DEC);
    Serial.print(':');
    Serial.print(future.second(), DEC);
    Serial.println();
    delay(1000);
}
```
But when I try to make the Timespan dependant of an IF condition like this :
 ```
#include "RTClib.h"
RTC_DS3231 rtc;
bool DST = true;   // or false, just for testing

void setup ()
{
  Serial.begin(57600);
  rtc.begin();
}

loop()
{
 DateTime now = rtc.now();
 // conditionnaly adding an hour
    if (DST = true)
       DateTime future (now + TimeSpan(0,1,0,0));
    else
       DateTime future (now + TimeSpan(0,0,0,0));

    Serial.print(future.year(), DEC);
    Serial.print('/');
    Serial.print(future.month(), DEC);
    Serial.print('/');
    Serial.print(future.day(), DEC);
    Serial.print(' ');
    Serial.print(future.hour(), DEC);
    Serial.print(':');
    Serial.print(future.minute(), DEC);
    Serial.print(':');
    Serial.print(future.second(), DEC);
    Serial.println();
    delay(1000);
}
```
I allways get this compilation error : 'future' was not declared in this scope.
Maybe I've missed something very simple but I'm stuck.

Any help would be appreciated.

Roland

edgar-bonet, 2024-04-09:
You have two declarations of `future`. It is not clear to the compiler which one you want to use further down. You should instead have a single declaration, like:

```c++
DateTime future;  // single declaration
if (DST)
    future = now + TimeSpan(0,1,0,0);
else
    future = now + TimeSpan(0,0,0,0);
```

Note that `loop()` should be declared to return `void`, as in `void loop() {...}`.

PS: This issue tracker is meant for tracking issues relative to the RTClib library. General programming questions like yours should not be asked here, but either on a forum (like the Arduino forum) or on a question/answer site (like arduino.stackexchange.com).

Rollmops67, 2024-04-09:
Oh my God, beginner error.

Thank you for your help, Edgar !

closed 2024-04-09 by Rollmops67