adafruit/RTClib #302: Cannot update to latest version 2.1.4

terrypin999, 2024-04-11:
I cannot update either with the Update button or by selecting Version 2.14
Why is version 2.12 not listed?
Shouldn't version 2.13 also be listed, even though it is installed?
See screenshot:
[image]

caternuson, 2024-04-11:
I think the 2.1.2 release is essentially non-existent due to an issue with not updating `library.properties` correctly. So 2.1.3 is 2.1.2. 

In the screenshot above, 2.1.3 is not in the drop down list since it is the currently installed version.
[image]

What happens when you try installing 2.1.4?

terrypin999, 2024-04-12:
Thank you.

As shown in my earlier screenshot, after using _Manage Libaries > Update_ on RTClib, it does not get implemented. Restarting IDE 1.8.19 shows the version apparently unchanged.

per1234, 2024-04-12:
@terrypin999 omitted some important information from this report:

https://forum.arduino.cc/t/cannot-update-rtclib/1246598

> I've just tried IDE 2.3.2 which is much more informative, telling me:
Failed to install library: 'RTClib:2.1.4'.
Error: 3 INVALID_ARGUMENT: The library RTClib has multiple installations:
>
>and listing several, all beginning with 'RTClib'.

@terrypin999 you also left out the part of the error message that gives instructions for how to solve the problem.

As explained by the error message from Arduino IDE 2.x, this problem occurs when there are multiple multiple installations of a given library. Libraries are identified by the `name` field in the `library.properties` metadata field. This means that it is possible to have multiple installations of a given library because, even though they must have different installation folder names to coexist in the file system, each library can have the same `name` field value in the metadata. Unless there is a malfunction, the Arduino IDE Library Manager will never produce multiple installations of a given library, but it can easily occur when the user installs libraries manually.

I'll provide instructions you can follow to reproduce the fault:

1. Use the Arduino IDE Library Manager to install version 2.1.3 of the "RTClib" library.
1. Click the following link to download a ZIP file of the library:
   https://github.com/adafruit/RTClib/archive/refs/tags/2.1.3.zip
1. Unzip the downloaded file in the `libraries` subfolder of your sketchbook folder.
   **ⓘ** You can also do this using the "[**Add .ZIP Library...**](https://docs.arduino.cc/software/ide-v1/tutorials/installing-libraries#importing-a-zip-library)" feature of Arduino IDE 1.x, but the feature in Arduino IDE 2.x is smart enough to not generate a redundant installation so you must do a fully manual installation to reproduce the fault if you are using 2.x.
1. Use the Arduino IDE Library Manager to update the "RTClib" library.

The update fails. If you are using Arduino IDE 2.x, you will see this error message like this in the "**Output**" panel:

```text
Failed to install library: 'RTClib:2.1.4'.
Error: 3 INVALID_ARGUMENT: The library RTClib has multiple installations:
- C:\Users\per\Documents\Arduino\libraries\RTClib
- C:\Users\per\Documents\Arduino\libraries\RTClib-master
Automatic library install can't be performed in this case, please manually remove all duplicates and retry.
```

If you are using Arduino IDE 1.x, the more vague message is shown:

```text
Library RTClib is already installed in: C:\Users\per\Documents\Arduino\libraries\RTClib
```

So @terrypin999's problem doesn't have anything to do with the codebase hosted in this library and the issue should be closed as off topic.

The failure of Arduino IDE 1.x to clearly communicate the problem to the user was certainly a defect (https://github.com/arduino/arduino-cli/issues/1871), but that defect has already been resolved to the extent possible in the years of development that have occurred since the time of the obsolete IDE version used by @terrypin999 so there is also no longer any problem in Arduino IDE.

caternuson, 2024-04-12:
FWIW, based on the name, this one was probably installed manually, via downloading repo zip:
```
- C:\Users\per\Documents\Arduino\libraries\RTClib-master
```