adafruit/RTClib #307: No support for DST

J-Brinkman, 2024-10-13:
A major part of Europe is using DST. However the library does not support DST. And implementing DST is not trivial. 
So please add support for DST. If not possible, than add the following code in one of the examples.

/* DST support */
/* CC BY-NC-SA Jeroen Brinkman */
/* Release 1: dd 23-10-2024 */

#include <RTClib.h>   // https://github.com/adafruit/rtclib 
RTC_DS3231 rtc;

#define USEDST  true // Use DST (true or false)

DateTime now;

DateTime getclock() { // Return the (DST adjusted) date and time
  
  DateTime n, b, e;
  
  n = rtc.now();
  b = DateTime(n.year(), 3, 31, 3, 0, 0); b = DateTime(n.year(), 3, 31 - b.dayOfTheWeek(), 1, 0, 0);
  e = DateTime(n.year(), 10, 31, 3, 0, 0); e = DateTime(n.year(), 10, 31 - e.dayOfTheWeek(), 3, 0, 0);

  if (USEDST && (n > b) && (n < e)) n = n + TimeSpan(0,1,0,0); // adjust standard time if within summertime
  return n;
}; 

void setclock(DateTime n) { // if the clock is set during summertime then adjust the clock to standard time

  if (USEDST && (rtc.now() != getclock())) n = n - TimeSpan(0,1,0,0); // if summertime then adjust to the standard time
  rtc.adjust(n); // Set the clock to standard time
}; 

void setup() {
  // initialise the rtc 
  rtc.begin();
  if (rtc.lostPower()) setclock(DateTime(F(__DATE__), F(__TIME__)));
}

void loop() {
  now = getclock(); // read the time from the RTC and adjust for DST
}

edgar-bonet, 2024-10-13:
> please add support for DST

This has already been discussed in the past, and the maintainers have stated that DST is out of scope for this library. See:

 * #144
 * #36

> CC BY-NC-SA

This library being free software (it's MIT-licensed), I wouldn't expect it to accept non-free contributions.

> ```c++
> b = DateTime(n.year(), 3, 31, 3, 0, 0); b = DateTime(n.year(), 3, 31 - b.dayOfTheWeek(), 1, 0, 0);
> e = DateTime(n.year(), 10, 31, 3, 0, 0); e = DateTime(n.year(), 10, 31 - e.dayOfTheWeek(), 3, 0, 0);
> ```

If I understand correctly, this rule applies only to [Central European Time (CET/CEST)][cet], not to the whole Europe.

[cet]: https://en.wikipedia.org/wiki/Central_European_Time

J-Brinkman, 2024-10-13:
Declaring it out of scope while The European Union (EU), including Bulgaria, France, Germany, Italy, Poland, and Spain. Most other European countries, including the United Kingdom, Norway, and Switzerland are using DST, is a bit short-sighted. Adding my example will solve this issue. I am willing to contribute without any (CC BY-NC-SA) restrictions.

dhalbert, 2024-10-13:
DST rules change over time, and vary by geography, as mentioned in #144. A complete database of timezone and DST rules, including dates in the past is described [here](https://data.iana.org/time-zones/tz-link.html), and is quite large. This library cannot hope to keep up with DST changes.

The simple `DateTime` class described here handles neither timezone nor DST adjustments. There are several  Arduino timezone-aware libraries that you could use to adjust a datetime from here. I found them by doing a websearch for `arduino datetime library with dst`.

I'm closing this as out of scope, as mentioned here and in the past.

closed 2024-10-13 by dhalbert

J-Brinkman, 2024-10-13:
Ignoring reoccurring feedback and a very simple solution, because you guys are living in the USA and therefore not facing the problems we have in Europe (and that is the complete EEC plus a number of extra countries) is quite remarkable and not very chic. Every customer in Europe therefore has to find a solution to deal with the DST and cannot use the official Arduino Library.  Jeroen Brinkman

dhalbert, 2024-10-13:
We have DST in the US and Canada also, so we are facing the same issues. There are some states that don't observe DST, next to other states in the same timezone that do.

edgar-bonet, 2024-10-13:
@J-Brinkman: You mentioned [Bulgaria][bul] and [the United Kingdom][uk], but neither of these countries is in the CET time zone: your code would not work over there!

Having a solution to handle DST would be a good thing, but it has to be more generic that your CET-only code. IMO, an example showing how to use RTClib together with, say, the [Timezone library][tz], would actually be relevant. At least more relevant than this CET-only solution.

> you guys are living in the USA

Nope.

[bul]: https://en.wikipedia.org/wiki/Time_in_Bulgaria
[uk]: https://en.wikipedia.org/wiki/Time_in_the_United_Kingdom
[tz]: https://github.com/JChristensen/Timezone

dhalbert, 2024-10-13:
You could add a simple API that turns on the addition of one hour or not. That would be user-controlled. It would not know about DST transition dates because they vary all over the world. The [Timezone library](https://github.com/JChristensen/Timezone) mentioned does not know dates: you give it some date info. Some other libraries that have DST support include https://github.com/FoxWhiskey/Time_DST, https://github.com/PaulStoffregen/Time, https://docs.arduino.cc/libraries/acetime/, and https://reference.arduino.cc/reference/en/libraries/dst-rtc/.

J-Brinkman, 2024-10-13:
Did you see my proposed solution?!? I asked for the addition of an example where the DST problem is solved. It took 2 procedures and 8 LOC's to implement European DST. I spend over a year to find this solution, because the approach is not trivial. And that is emphasized by all the complex libraries the internet refers to. Other countries may have other DST rules. An example will provide a big help to someone to adapt these procedures to local DST rules. At least I would really liked to have had them upfront. My solution can be found here: https://cdn.hackaday.io/files/1984388489971392/DST%20support.ino

dhalbert, 2024-10-13:
Aha, I missed that it was an **example**, as opposed to changing the library code. We'd be happy to entertain a pull request to https://github.com/adafruit/RTClib/tree/master/examples. That is no problem.

J-Brinkman, 2024-10-13:
> @J-Brinkman: You mentioned [Bulgaria](https://en.wikipedia.org/wiki/Time_in_Bulgaria) and [the United Kingdom](https://en.wikipedia.org/wiki/Time_in_the_United_Kingdom), but neither of these countries is in the CET time zone: your code would not work over there!
> 
> Having a solution to handle DST would be a good thing, but it has to be more generic that your CET-only code. IMO, an example showing how to use RTClib together with, say, the [Timezone library](https://github.com/JChristensen/Timezone), would actually be relevant. At least more relevant than this CET-only solution.
> 
> > you guys are living in the USA
> 
> Nope.

Oh yes it would. work over in the UK and Bulgaria. Last sunday in march and last sunday in october. Please look at my proposed code:

J-Brinkman, 2024-10-13:
> @J-Brinkman: You mentioned [Bulgaria](https://en.wikipedia.org/wiki/Time_in_Bulgaria) and [the United Kingdom](https://en.wikipedia.org/wiki/Time_in_the_United_Kingdom), but neither of these countries is in the CET time zone: your code would not work over there!

Oh yes it would. Both Bulgaria and the UK have the same rules as the EU: last sunday in march and october. The hour can differ per country. But that can easily be adapted in my proposed code. And by the way: I don't care if my clock is set at 1:00, 2:00 or 3:00  in the middle of the night to DST. That is of minor importance. You don't need a perfect solution, if a good solution will do.