adafruit/RTClib #311: API of DS3231 doesn't have a way to check that an alarm is enabled

Spartrap, 2025-03-14:
Hello,
Concerning the Alarm features of the DS3231,
the API contains:

  void disableAlarm(uint8_t alarm_num);

but there's no obvious corresponding API for:

 a) checking if an alarm is enabled
 b) enabling an alarm

It would be extremely handy to have these.

Greetings,
Germain

edgar-bonet, 2025-03-14:
The API of the `RTC_DS3231` class contains the following alarm-related methods:

```c++
bool setAlarm1(const DateTime &dt, Ds3231Alarm1Mode alarm_mode);
bool setAlarm2(const DateTime &dt, Ds3231Alarm2Mode alarm_mode);
DateTime getAlarm1();
DateTime getAlarm2();
Ds3231Alarm1Mode getAlarm1Mode();
Ds3231Alarm2Mode getAlarm2Mode();
void disableAlarm(uint8_t alarm_num);
void clearAlarm(uint8_t alarm_num);
bool alarmFired(uint8_t alarm_num);
```

See the [online documentation of the class][doc].

[doc]: https://adafruit.github.io/RTClib/html/class_r_t_c___d_s3231.html

Spartrap, 2025-03-15:
Yes indeed. And at the semantical level, it strikes me that none of those methods allow to check if an alarm is "disabled", or to "enable" back an alarm after the use of disableAlarm().

edgar-bonet, 2025-03-15:
@Spartrap: Oh, you are right! There is no way to check whether an alarm is disabled.

Note, however, that the methods `setAlarm1()` and `setAlarm2()` enable the respective alarms. It may be a bit annoying that you have to repeat the alarm settings (if the alarm has already been set at the desired time and mode), but at least you have this way of enabling again a disabled alarm.