adafruit/RTClib #312: 2 Seconds passing after one second on DS1307

bentoemanuel, 2025-06-04:
I connected the RTC module, connected to the arduino via the I2C port and executed the example code, adjusting the time for today, however the Serial Monitor displays something close to this:
`03/06/2025 12:30:05 -> 12:30:05
03/06/2025 12:30:06 -> 12:30:07
03/06/2025 12:30:07 -> 12:30:09
03/06/2025 12:30:08 -> 12:30:11 `

This issue is persistent with whichever code I try to execute using the RTC, always doubling the amount of seconds that have been elapsed.

My workaround was to store the current date times two when adjusting the time on the RTC and when reading to divide it by two, but this solution would also divide the clocks range to half the unix time range

edgar-bonet, 2025-06-04:
Could you please show the example code you are using?

bentoemanuel, 2025-06-04:
Here it is:

`#include <Wire.h>
#include "RTClib.h"

RTC_DS1307 rtc;
  
char diasDaSemana[7][12] = {"Domingo", "Segunda", "Terca", "Quarta", "Quinta", "Sexta", "Sabado"};

void setup () {
  Serial.begin(9600);                      
  if (! rtc.begin()) {                      
    Serial.println("RTC NAO INICIALIZADO");   
    while (1);                            
  }
  //rtc.adjust(DateTime(1748955300*2));
  delay(100);                             
}

void loop () {
  DateTime agora = (rtc.now().unixtime()/2);
  Serial.print("Data: ");
  Serial.print(agora.day(), DEC);                      
  Serial.print('/');                              
  Serial.print(agora.month(), DEC);                       
  Serial.print('/');                                     
  Serial.print(agora.year(), DEC);                      
  Serial.print(" / Dia da semana: ");                    
  Serial.print(diasDaSemana[agora.dayOfTheWeek()]);  
  Serial.print(" / Horas: ");                          
  Serial.print(agora.hour(), DEC);                    
  Serial.print(':');                                 
  Serial.print(agora.minute(), DEC);                 
  Serial.print(':');                                 
  Serial.print(agora.second(), DEC);                   
  Serial.println();                                    
  delay(1000);                                          
} `

As I have mentioned, I am storing the time as double the actual time because of this bug.

edgar-bonet, 2025-06-04:
@bentoemanuel: What does the following program print?

```c++
#include <Wire.h>
#include <RTClib.h>

RTC_DS1307 rtc;

void setup () {
    Serial.begin(9600);
    if (!rtc.begin()) abort();
    rtc.adjust(DateTime(2025, 6, 4, 12, 0, 0));
}

void loop () {
    Serial.println(rtc.now().timestamp());
    delay(1000);
}
```

edgar-bonet, 2025-06-05:
@bentoemanuel: This is really weird. Looks like a hardware issue to me. Are you sure you connected a 32.768&nbsp;kHz quartz crystal? Could it be oscillating at an overtone mode?

bentoemanuel, 2025-06-13:
I couldn't fix this RTC so I bought another one, from what I have tried,
-> SQW pin on the module is OK for all frequencies.
-> Device is communicating to Arduino, produces the same output on all boards I tried.
-> NVRAM reads and writes OK.
-> No pins are shorting.
-> No diodes or resistors are shorting.
-> Battery is charged and working.
From what I've analysed I suspect the problem is in the EEPROM chip but I am not sure. I will be closing this issue because I couldn't find a solution.

closed 2025-06-13 by bentoemanuel